bbcmicrobit/micropython #48: "TypeError: Unicode-objects must be encoded before hashing" when running: yt up

amos1969, 2015-10-20:
Hi 

I'm working through the instructions to compile MicroPython for the micro:bit, when I run **yt up** I get the following:

info: get versions for bbc-microbit-classic-gcc-nosd
info: get versions for mbed-gcc
info: get versions for mbed-classic
info: get versions for ble
Traceback (most recent call last):
  File "/usr/local/bin/yt", line 4, in <module>
    yotta.main()
  File "/usr/local/lib/python3.5/site-packages/yotta/main.py", line 239, in main
    status = args.command(args, following_args)
  File "/usr/local/lib/python3.5/site-packages/yotta/update.py", line 28, in execCommand
    updateDeps(args)
  File "/usr/local/lib/python3.5/site-packages/yotta/update.py", line 58, in updateDeps
    test = update_test_deps
  File "/usr/local/lib/python3.5/site-packages/yotta/lib/component.py", line 554, in satisfyDependenciesRecursive
    test = test
  File "/usr/local/lib/python3.5/site-packages/yotta/lib/component.py", line 349, in __getDependenciesRecursiveWithProvider
    test = test
  File "/usr/local/lib/python3.5/site-packages/yotta/lib/component.py", line 253, in __getDependenciesWithProvider
    return (OrderedDict([((d and d.getName()) or specs[i].name, d) for i, d in enumerate(dependencies)]), errors)
  File "/usr/local/lib/python3.5/site-packages/yotta/lib/component.py", line 253, in <listcomp>
    return (OrderedDict([((d and d.getName()) or specs[i].name, d) for i, d in enumerate(dependencies)]), errors)
  File "/usr/local/lib/python3.5/site-packages/yotta/lib/component.py", line 231, in satisfyDep
    self.getName()
  File "/usr/local/lib/python3.5/site-packages/yotta/lib/component.py", line 540, in provider
    r = access.satisfyVersionByInstalling(dspec.name, dspec.version_req, self.modulesPath())
  File "/usr/local/lib/python3.5/site-packages/yotta/lib/access.py", line 294, in satisfyVersionByInstalling
    v = latestSuitableVersion(name, version_required, _registryNamespaceForType(type))
  File "/usr/local/lib/python3.5/site-packages/yotta/lib/access.py", line 145, in latestSuitableVersion
    remote_component.availableTags(),
  File "/usr/local/lib/python3.5/site-packages/yotta/lib/github_access.py", line 249, in availableTags
    ) for t in self._getTags()
  File "/usr/local/lib/python3.5/site-packages/yotta/lib/github_access.py", line 249, in <listcomp>
    ) for t in self._getTags()
  File "/usr/local/lib/python3.5/site-packages/yotta/lib/github_access.py", line 163, in _createCacheKey
    h.update('this is the _createCacheKey seed')
TypeError: Unicode-objects must be encoded before hashing

ntoll, 2015-10-20:
Yotta is Python 2.7 only (at the moment).

Yotta is ARM's build tool so we can't do anything at the moment about this since it's upstream. :-(

Submit a feature request here: https://github.com/ARMmbed/yotta ;-)

closed 2015-10-20 by ntoll

amos1969, 2015-10-20:
Hi Nicholas

In the instructions it says to do **pip3 install yotta**, which is what will have caused the issue I guess. I'm going to go and pip install it and see if that alleviates the problem.

Cheers
Dave

amos1969, 2015-10-20:
Uninstalling yotta (pip3 uninstall yotta) and then reinstalling using pip install yotta fixed this.