bbcmicrobit/micropython #49: error: could not locate github component "lancaster-university/microbit-dal"

amos1969, 2015-10-20:
Hi 

Running **yt up** (using Python 2.7 on Linux Mint 17.2) I now get the error:

error: could not locate github component "lancaster-university/microbit-dal", either the name is misspelt, you do not have access to it, or it does not exist

Looking at the lancaster-university page on github I can't see anything similar.

ntoll, 2015-10-20:
This is the device access layer (DAL) that is still under the NDA.

If you would like access you need to have signed the NDA (@amos1969 - I believe you're already covered) and then Lancaster Uni can add you.

willingc, 2015-11-02:
@ntoll Does the NDA for the DAL still apply? Getting the error of not found when running ``yt up``

ntoll, 2015-11-02:
'fraid so (we're working on it). In the meantime, as a director of the PSF I believe you're covered by the NDA (i.e. just don't share the DAL). Ergo, I can ask Joe and/or James to add you to the repos. GitHub will ping you when they've done it (they're usually very quick). :-)

willingc, 2015-11-02:
Sounds good. Thanks @ntoll.

deshipu, 2015-11-24:
I would also want access to that repository, for the github user ``armmbedyotta`` (I can't use my normal user, because yotta wants access to private repositories), if it's possible?

dpgeorge, 2015-11-24:
> I can't use my normal user, because yotta wants access to private repositories

I had similar concerns, but was convinced that giving yotta access is no less secure than, eg, logging into GitHub with Firefox.

ntoll, 2015-11-24:
OK... I've requested you have access. You should get an email from GitHub when you're in the "club". ;-)

dpgeorge, 2016-04-09:
Now available.

closed 2016-04-09 by dpgeorge