bbcmicrobit/micropython #51: Include package list for Fedora?

rjw57, 2015-10-20:
I think that Fedora (should you wish to play the "documenting multiple distros" game) has the appropriate GCC available in it's repos as ``arm-none-eabi-gcc-cs`` and ``arm-none-eabi-gcc-cs-c++`` meaning that the packages required for can be installed via:
```console
$ sudo dnf install cmake ninja-build srecord arm-none-eabi-gcc-cs arm-none-eabi-gcc-cs
```
In addition, the `` python3-cryptography`` and ``python3-cffi`` packages are needed to allow the ``pip2 install yotta`` to complete without any native code compilation steps. (Yotta GitHub checkouts appears to fail on Python 3 on my machine but that's a separate issue.)

I can't fully test this locally since a) I'm not NDA-blessed meaning that I can't clone the DAL and b) although being in Cambridge I'm probably minute away from the nearest micro:bit, I don't have one to hand :). Otherwise, I'd have opened a PR for the README.

ncoghlan, 2016-01-17:
Folks looking for an environment usable on Fedora may also want to take a look at my Docker based environment: https://github.com/ncoghlan/microbit/tree/master/devenv

Angus Tomas also provided some additional details on getting a native Fedora environment up and running as part of the micro:bit world tour: https://microworldtour.github.io/microbit/monifa.html

(Both approaches still require NDA access to the DAL components though)

ntoll, 2016-01-17:
@ncoghlan myself and various other partners are pushing the BBC to allow for the DAL to be opened up. I'll keep you posted.

Ultimately, the partnership agreement says that the software should be released as open source when the devices are delivered to the kids in school. That won't be too far into the future as far as I can tell.