bbcmicrobit/micropython #59: Document the correct way to callibrate the compass

ntoll, 2015-10-30:
Apparently, the correct way to calibrate the compass is to turn it flat on the spot (without tilting it otherwise that will add inaccuracies). We should document this in an obvious place.

willingc, 2015-10-30:
@ntoll Perhaps the easiest way to document initially is in an introductory docstring block:

```
"""
    compass.py
    ~~~~~~~~~

    Creates a compass. To calibrate the compass, lie it flat and spin it horizontally. For 
    best results during calibration, try not to tilt the compass up and down.

"""
```

ntoll, 2015-10-30:
+1 - also there should be something in the api so `help(microbit.compass.callibrate)` returns something useful that an 11yo kid would understand.

ntoll, 2015-11-30:
Docstring added, the `help` related text still doesn't reference how to calibrate it. This will be fixed by #125.

PR #125 (2015-11-30): Add calibration guidance to help text.
A very simple addition of calibration instructions so we can close #59 for 0.9 release.

Please merge ASAPs. :shipit:
closed 2015-11-30 by dpgeorge