bbcmicrobit/micropython #63: import love no longer works

ntoll, 2015-11-02:
To recreate, in the REPL type, `import love` and the same incorrect random pattern appears and freezes the device. I guess this is because `Image` has changed. :-/

ntoll, 2015-11-02:
OK... by the looks of it, the heart is no longer beating, but glowing/dimming to represent the beat. My C++ knowledge is very limited so I'm not sure how further to proceed. AFAICT  `microbit_image_dim` is returning something that barfs `display.print`.

willingc, 2015-11-02:
:broken_heart:

dpgeorge, 2015-11-03:
Fixed.

dpgeorge, 2015-11-03:
@markshannon sorry, only just saw that you assigned yourself to this issue!  Fix is here: 72c86d7800def5b03ea946cb02bcc7974e5d2f17 .  Of course, what else could it be? :)

closed 2015-11-03 by dpgeorge

markshannon, 2015-11-03:
Thanks. It's horrible having to copy the shape of the C++ class. Reminds me of this:
http://steve-yegge.blogspot.co.uk/2010/07/wikileaks-to-leak-5000-open-source-java.html