bbcmicrobit/micropython #64: Add doxygen developer docs

willingc, 2015-11-02:
As I was walking through the code base, I created a [Doxyfile](https://github.com/willingc/micropython-1/blob/3eddb8636b750839bed8e344057a4501e2a7241a/Doxyfile) to use with [doxygen](http://www.stack.nl/~dimitri/doxygen/) for creating documentation overviews of the code and its structure. I thought others might find it helpful or we may wish to add to the documentation.

dpgeorge, 2015-11-02:
I'm not a big fan of Doxygen, but other people may be :)  I guess it's only useful to view docs from the DAL and/or mbed?

ntoll, 2015-11-30:
@willingc any progress on this or can we shelve this "feature" for 0.9?

willingc, 2015-11-30:
I wasn't planning on moving forward with this. It may be useful down the road for docs. For now, I will close.

closed 2015-11-30 by willingc