bbcmicrobit/micropython #68: Ensure random() is truely random

ntoll, 2015-11-07:
There is an issue with the DAL implementation of `random()`. We should supply an up-stream patch to fix this.

whaleygeek, 2015-11-12:
Yes, it's not very random at all at the moment. My Dalek game always (reliably) shoots me on the first two attempts, implying to me that the random is not correctly seeded?

ntoll, 2015-11-12:
@whaleygeek hey hey... good to have your input! It turns out that @dpgeorge found a bug in the DAL implementation of the RNG but it's not yet been fixed.

ntoll, 2015-11-30:
Appears to be fixed upstream here: https://github.com/lancaster-university/microbit-dal/pull/46

closed 2015-11-30 by ntoll

dpgeorge, 2015-11-30:
I have verified that the patch works and random is now random (enough)!