bbcmicrobit/micropython #71: Add Bluetooth Low Energy (BLE) capabilities to MicroPython

ntoll, 2015-11-07:
Because the BLE stack takes up too much RAM for MicroPython to be able to compile scripts to byte code Damien has a plan:

* If the script requires BLE when the device restarts immediately after flashing it will compile the Python script to byte code (without BLE on) and store the result in flash.
* It will then restart the device and execute the byte code with BLE on.
* The REPL will not be available in this mode.

willingc, 2015-11-08:
@dpgeorge :heart:

dpgeorge, 2015-11-08:
We will try our best to make BLE work!

ntoll, 2015-11-30:
@dpgeorge shall we bump this to 1.1???

dpgeorge, 2015-11-30:
Yes, I think we should.  Otherwise v1.0 might take some time.

zeth, 2016-03-10:
I know little about bluetooth hitherto, but:

"the Bluetooth profile provides access to the key hardware features such as the buttons, accelerometer, magnetometer, pins and LEDs."

Stupid question but could we make a smaller bluetooth profile? I.e. instead of exposing a massive API, just have the ability to transmit ints? What the ints mean could then be dealt with by the user in Python code.

http://www.electronicsweekly.com/uncategorised/interview-making-bbc-microbit-bluetooth-smart-2015-07/
https://www.bluetooth.com/download-developer-studio

dpgeorge, 2016-03-10:
@zeth unfortunatelly there is a lot of code and RAM needed just to transmit ints over BLE.  Once you have this capability it's not much more to add buttons, accel, etc.

finneyj, 2016-03-10:
@zeth, @dpgeorge 

Would be great to see funcitonality in Pythin too, but this would be very tricky i think. We've been working with the Nordic BLE stack for sometime now to help support the other languages, and it really is a VERY memory intensive (and a rather invasive beast). The stack needs about 100k of FLASH, statically allocated... and that's for the smallest version of the 'soft device' stack currnently available chip. In addition, the stack also statically reserves 8K of SRAM before it starts. It also need another 1.5K SRAM to support its SDKs, and consumes up to 2K more on the stack when any deep interrupts fire. 

All in all, it costs about 100K flash and up to 11.5K SRAM. Also, BLE isn't really a bytestream protocol  - it provides REST like services to connected devices. Each of these cost additional RAM in the stack too (about 180 bytes per characteristic - think REST variables).

For Blocks/TD/JS we bring up a minimal set to allow over the air programming and an extension of the MessageBus (which does pretty much what @zeth suggests - sends a few bytes that can be used to trigger events)... The other services are only brought up on demand.

Totally agree that services should be more tailored to the app to support IoT scenarios though - a profile as we've created thus far is the current common practice BLE communities according to our friends at Bluetooth SIG, so provides a start point only IMHO...

zeth, 2016-03-10:
@dpgeorge Thanks for answering. Never mind. I guess if kids want to make two player games they can use crocodile leads or whatever which is covered under https://github.com/bbcmicrobit/micropython/issues/37

dpgeorge, 2016-03-10:
There is an alternative "radio" protocol which is not BLE but which would allow microbit's to communicate.

carlosperate, 2016-03-10:
Is that the Microsoft blocks/touchdevelop are currently using? I was wondering why it was named like that.

finneyj, 2016-03-10:
@carlosperate Yep - the code is in MicroBitRadio.cpp and MicroBitRadioDatagram.cpp in the microbit-dal repo if you're interested to see how this works.

This is a much simpler protocol (just allows send/receive of data frames).

zeth, 2016-03-10:
For the use case of Pong/Football/Battleships/etc that radio protocol sounds perfect. This is my first week of microbittery so I don't know where is the microbit-dal but I found this: 

https://github.com/Microsoft/yelm/blob/master/libs/microbit-radio/radio.ts

Something approximating that would be great.

finneyj, 2016-03-10:
@zeth oh, OK - sorry. microbit-dal is a C/C++ layer that provides a set of device drivers and some runtime functionality to the languages that cross-compile for micro:bit (including Blocks, Touch Develop etc). microPython uses a bit of this too. @jamesadevine and myself here at Lancaster are We're currently looking at refactoring this layer to be more component based to make this easier going forward for languages that have their own runtime environments already (such as MicroPython, espruino etc).

Yes, the library ypou mention is the TD wrapper around the microbit-dal implementation. The repo is here, but sadly we're not quite permitted to open source just yet (due to contract with BBC): https://github.com/lancaster-university/microbit-dal

If you've been building micropython, you should already have access though.

This was also mentioned in a related issue, and please be aware that the BBC haven't quite finished putting this functionality through their data protection / child safeguarding processes yet, so we can't issue any public APIs containing this yet... :-(  but hopefully won't be long before we can.

markshannon, 2016-06-21:
Are we going to scrap this in favour of the simple radio module?
I'd like to. I thing the simple radio has much more value from a teaching perspective and it uses sooooo much less RAM.

ntoll, 2016-06-21:
@markshannon totally agree. I think the lack of RAM hobbles the use of BLE on the device. Until this changes I think the radio module, as we've been discussing elsewhere, fulfils all the educational requirements, is *a lot simpler* to use and, importantly for us, doesn't require 12k of RAM.

dpgeorge, 2016-06-24:
Even with enough RAM (eg on a 32k device) it's still going to be hard to support BLE because of the flash requirements: the soft device needs 112k, leaving 144k for application.  We'd need to strip out a lot of Python features to get the uPy firmware under 144k.

ntoll, 2016-06-24:
I see this as an intractable issue as the hardware specifications currently stand. I'm going to close this issue on the understanding that if circumstances change, someone with the time and inclination can create a PR that adds BLE support for us to test and merge.

closed 2016-06-24 by ntoll

Navis-Raven, 2018-07-27:
Hi, there, BBC micro:bit is a real good cardboard not only for students but also for makers, and is far more better than Arduino UNO (since it includes many programming languages: C++, Pascal, Scratch, µPython). And one of the greath thing is that it is one of the few cardboards that are programmable in µPython. So it is too bad that µPython does not include also the BLE.

I didn't understand why by just replacing the actual nRF51822 of 16k by a 32k one will not allow the µPython support the Bluetooth LE ?

I have also some other questions, in order to include BLE:
Is µPython optimisable by calling some function in C/C++ language instead of Python ? It might free many ressources isn't it ?
In this context, in C++ BLE stack, will it change something if it is rewrote in just C language (with possibly some blocks in ASM) ?

orklann, 2018-08-27:
Is it possible to build byte code outside of Microbit itself, so that we can build BLE module for MicroPython on our PC/MAC, and then flash the byte code into Microbit?

kevinjwalters, 2020-10-13:
@Navis-Raven Someone heard you: https://microbit.org/new-microbit/ ([nRF52833 based](https://www.nordicsemi.com/News/2020/10/Nordic-powered-next-generation-microbit-to-support-AI-and-machine-learning-applications).

Navis-Raven, 2020-10-14:
@kevinjwalters thank you,  good to hear the new BBC:microbit is far more better then the previous one, and far more better than Arduino cardboards. The actual cardboard could even be used in industry to make easiest prototypes

Navis-Raven, 2020-10-14:
@kevinjwalters  but I didn't found anything on BBC website about Bluetooth, do you have a weblink about this ?

kevinjwalters, 2020-10-14:
There's a good table in https://tech.microbit.org/latest-revision/. The hardware is capable of BLE 5.0. The limitation with MicroPython and BLE 4 was trying to cram everything in on the humble nRF51822, I'd imagine that goes away with increased ram and flash but those in the know can chime in and perhaps already have plans.

Navis-Raven, 2020-10-14:
@kevinjwalters  thank you

penguintutor, 2020-10-17:
I'm looking forward to the V2. Hopefully the extra memory will make it possible to have Bluetooth working properly on the MicroBit.

Navis-Raven, 2020-10-18:
@ntoll I think you can reopen the issue now.