bbcmicrobit/micropython #73: Allow persistent storage in flash

ntoll, 2015-11-07:
It should be possible to use flash based persistent storage to squirrel away "stuff" between resets. Perhaps a very simple key/value store that looks like a dictionary to the end user would suffice?

dpgeorge, 2015-11-07:
Main thing to know for this is: what is the page-erase size on the nrf51?  That will dictate how it's done.  Maybe we also want some kind of simple wear-levelling.

whaleygeek, 2015-11-12:
From the NRF51 datasheet, the device will support 20,000 erase cycles. The smallest write size is a 32 bit word. Page (and therefore erase) size is 1024 bytes at a time. The MPU supports protected regions in 4kB chunks.

ntoll, 2015-11-30:
@dpgeorge did I imagine this, or has this been completed..?

dpgeorge, 2015-11-30:
It's not implemented, not even in the DAL.

ntoll, 2015-11-30:
OK. I must have mis-remembered a commit message.

markshannon, 2016-05-13:
Superseded by https://github.com/bbcmicrobit/micropython/issues/246

closed 2016-05-13 by markshannon