bbcmicrobit/micropython #74: Create built in image effects.

ntoll, 2015-11-07:
Pass in a single image and get back a list of images to animate a referenced effect:

 - [ ] sparkle - active pixels in the original image are dimmed / brightened in a random fashion.
 - [ ] throb - the throbbing effect seen when you `import love`
 * Wipe effects - the image fades in from the specified direction, until all the pixels are fully on:
   - [ ] wipe_left
   - [ ] wipe_right
   - [ ] wipe_up
   - [ ] wipe_down
 - [ ] radar - a circular motion is created using the active pixels and a rotating "bar" of bright pixels that gradually dim to darkness.
 * Fade in/out - the whole image fades in / out to / from dark to bright:
   - [ ] fade_in
   - [ ] fade_out

Perhaps these functions could live under `microbit.display.fx`..? E.g.

```
>>> twinkle_happy = display.fx.sparkle(Image.HAPPY)
>>> twinkle_happy
[ ... a shortish list of Image objects ... ] 
>>> display.animate(twinkle_happy, 50)
```

dpgeorge, 2015-11-07:
> Perhaps these functions could live under microbit.display.fx..?

Or just be methods of an Image instance, eg:
```python
>>> twinkle_happy = Image.HAPPY.sparkle()
```

markshannon, 2015-11-11:
I would suggest that each "effect" function should take a single image and return an iterators of images.
They could be methods on images, but iterators are not the simplest of concepts, so its best to wrap them in a utility functions.
Something like:
`display.FX.sparkle(img)` which would call `display.animate(_FX.sparkle(img))`

deshipu, 2015-11-11:
So I have this idea for generalizing ``display.print()`` (or ``display.show()`` if it gets renamed).

Right now, you can do ``display.print("text", delay=100)`` or ``display.print(image)``. It would be logical to me to use the same function for the second form of ``display.animate``, that is, ``display.show([image1, image2, image3], delay=100)``. That would require adding a special case for iterables in there. But that would also allow us to make "animation" objects, that are basically iterables of images. You could create those, passing any images or effect parameters to them, and they would produce the animation frames when you iterate over them, using the iterator protocol.

If you did that, you could have:

    twinkle_happy = fx.Sparkle(Image.HAPPY)
    display.show(twinkle_happy, delay=200)

That would also let you combine the animations, if they accepterd iterables as input.

An additional thing (not sure if that makes sense), we could also look for a ``__show__`` attribute on the objects that are not iterables, and iterate over that if it exists. That would let you to call ``display.show()`` on user objects, similar to how you can print them using ``__repr__``.

deshipu, 2015-11-11:
I think that effects should rather be objects using the iterator protocol and generating the frames on the fly, than functions returning lists (that saves memory).

It would be great if they could accept both a single image (in which case they would use it for the base of each frame), or an iterable of images (in which case the effect would be added to each image from the iterable) -- that would let us compose animations.

The ``display.animate`` function duplicates what ``display.print`` already does for text, maybe those two could be merged into a single ``display.show`` that takes either an image or an iterable of images or characters.

Not sure if that is a good idea, but there could also be a "magical" attribute that ``display.show`` looks for, similar to how ``__repr__`` works for print, which would allow making any user object "displayable".

markshannon, 2015-11-12:
On 11/11/15 09:46, Radomir Dopieralski wrote:
> I think that effects should rather be objects using the iterator
> protocol and generating the frames on the fly, than functions returning
> lists (that saves memory).
Indeed, I think we are in agreement.

>
> It would be great if they could accept both a single image (in which
> case they would use it for the base of each frame), or an iterable of
> images (in which case the effect would be added to each image from the
> iterable) -- that would let us compose animations.
Here we disagree. I think it would make the interface overly complex and 
potentially confusing. You can already compose iterables and functions:
`display.show(chain(map(fx.sparkle, list_of_images)))`
(A more readable form is left as an exercise :) )

>
> The |display.animate| function duplicates what |display.print| already
> does for text, maybe those two could be merged into a single
> |display.show| that takes either an image or an iterable of images or
> characters.
Indeed. In fact that is what `scroll` and `animate` do already behind 
the scenes; they create an iterator of images which is then passed to 
`display.print/show`

>
> Not sure if that is a good idea, but there could also be a "magical"
> attribute that |display.show| looks for, similar to how |__repr__| works
> for print, which would allow making any user object "displayable".
`__image__` presumably?

deshipu, 2015-11-12:
> Here we disagree. I think it would make the interface overly complex and
potentially confusing. You can already compose iterables and functions:
`display.show(chain(map(fx.sparkle, list_of_images)))`
(A more readable form is left as an exercise :) )

That doesn't do the same thing. It would play an animation of each image sparkling, one after the other.
However, if fx.sparkle would accept an iterator and then use each image as the base for each of its frames, you could have something like:

    display.show(fx.shake(fx.sparkle(Image.HEART))

which would display an animation of a heart *both* shaking *and* sparkling *at the same time*. You can't do that in any other way (except for some trivial cases, where you could maybe use the + operator to overlay one list of images on another, but most animations are more than just adding an image on top).

markshannon, 2015-11-22:
I don't see how effects could be composable in the way you describe.
What would `fx.wipe_left(fx.wipe_right(some_image))` do, and how would it work?

deshipu, 2015-11-22:
Of course it would work. It would wipe the image from both sides at the same time, narrowing it, so to say.

deshipu, 2015-11-22:
For the effects that cannot take a different image for the base of each frame, they could fall back to just using the first frame. But most effects are easily composable by just applying all previous steps to the next base image.

markshannon, 2015-11-22:
Here is a simple implementation of `wipe_left` implementing the `Image -> List(Image)` interface. 
``python
WIPE_LEFT = Image("99999876543210000:"*5)
def wipe_left(img):

How would you implement the `List(Image) -> List(Image)`?

deshipu, 2015-11-22:
WIPE_LEFT = Image("99999876543210000:"*5)
    def wipe_left(imgs):
        for i in range(12, -1, -1):
              yield WIPE_LEFT.crop(i, 0, 5, 5) * next(imgs)
        yield from imgs

deshipu, 2015-11-22:
Of course you need to have a check if the thing is iterable.

markshannon, 2015-11-22:
Presumable you would just convert a single image to a sequence of the same image being repeated?
Would all effects need to be infinite sequences?

markshannon, 2015-11-22:
`being repeated` -> repeated for ever`

deshipu, 2015-11-22:
Well, this is something to be discussed -- how to handle sequences of different lengths. As you see, here I just apply the effect to the first 12 frames, and then return all the rest of frames unchanged. It may make sense to do something else, depending on the effect. For instance, for a fade-out effect, it may make sense to return empty frames.

I don't think returning infinite sequences is wise, since you still need to display it -- and the display function needs to know when to stop.

Note also, that if the ``imgs`` iterator at any moment ends, raising ``StopIteration``, that will also end the ``wipe_left`` iterator, so the resulting sequence is as most as long as the original sequence. This may not always be true -- you may have effects such as "slow motion" or "fast forward", that actually add or remove frames. For other effects, it would probably make sense to specify with an additional parameter how many frames the effect should take, and maybe even how many frames should be skipped from the beginning?