bbcmicrobit/micropython #76: API documentation

ntoll, 2015-11-07:
We need ReadTheDocs documentation for all the aspects of the API to the same quality as that for the music namespace found here: 

https://github.com/bbcmicrobit/micropython/blob/master/docs/music_api.rst

deshipu, 2015-11-07:
I started with the compass module: https://github.com/bbcmicrobit/micropython/pull/78

deshipu, 2015-11-08:
￼ Left: i2c, Image, display, uart

deshipu, 2015-11-09:
I think that at this point we have pull requests for all the required docs. Some of them require more work, of course.

willingc, 2015-11-09:
@deshipu That's wonderful! Great effort.

deshipu, 2015-11-09:
oops, Image is still missing

carlosperate, 2015-11-13:
Perhaps this has already been defined in a different discussion. But for this type of documentation, what is the assumed technical knowledge of the target audience? Do all electronic/software concepts have to be explain (within reason), or can we assume a minimum technical knowledge baseline?

For example, in the [buttons documentation](https://github.com/bbcmicrobit/micropython/pull/79#issuecomment-156419646) it discussed that the reader could be familiar with the singleton concept, so can we also assume they are familiar with general embedded communication protocols? PWM or any other peripherals basic operation?

dpgeorge, 2015-11-13:
There are going to be many sources of "documentation", eg: that provided by help(obj), the stuff in the docs/ directory, tutorials and learning resources, etc.  I think the stuff in docs/ should be the official formal spec providing authoritative information.  But it should also be able to be read and understood by an intelligent 11yo.

ntoll, 2015-11-13:
In my minds eye these are the authoritative technical documents for the API. Ergo, we should be able to assume some level of technical proficiency although I agree with @dpgeorge that any reasonably intelligent 11yo who knows how to use Google to search for terms will be able to comprehend these docs. 

The built in `help(obj)` documentation, in contrast, is specifically aimed at kids with a reading age of no more than 11.

willingc, 2015-11-13:
We should anticipate that the files in the ``docs/`` directory will be read by interested 11 yo, as @dpgeorge mentions, as well as teachers and other adults. It would be great to set the tone and level of the documentation similar to the excellent work by Adafruit's tutorials. Clarity is the top goal, followed by minimizing jargon (and defining as needed), and helpful tips to connect readily available hw devices to the micro:bit. Part of what will make the micro:bit fun for students is the ability to connect other things to it.

deshipu, 2015-11-16:
I agree that it would be absolutely marvelous if the documentation was written in simple language and friendly tone, explained all technical terms used and generally be suitable for just giving the kids to read directly without any work from the teachers. In a perfect world that would be the case. However, there is a cost to that, and I'm not sure we realistically can afford it.

You see, we would need an army of people working on that long hours, who would not only be experienced engineers and proficient programmers, but also excellent writers and experienced teachers. While there may be some such people in the world, I'm not sure they would have the time to devote to this particular task, and that this would be actually a good use of their superhuman skills.

I know enough about electronics to be dangerous with a soldering iron, know C enough to tell a function declaration from a comment, and I know English enough to order a pint of ale, provided my hands are free to gesture. And I'm completely miserable around children. I'm certainly not able to write such documentation that you request.

I think the absolute necessity is to have one source of information that is 1) accurate, 2) complete and 3) up to date. This will allow the people who have more experience with teaching and writing to take it and, based on the information contained there, create the actual lesson materials needed by the children and their fellow teachers. Without it, all they can do is sit and twiddle their thumbs. So I think that this should be the priority.

Of course, it's good if we can also make it simple, friendly and comprehensive at the same time, but there is a small problem with this. If the docs grow too elaborate because of that, it will be difficult to update them as things change during the board's continued development. Ideally those updates would happen in the same commits that change things, or closely follow. But if it's too much work and requires special skills, that will be delayed and we will have documentation that is out of sync with reality, that is, we will have a huge pile of lies, that will in turn propagate into all other materials.

I believe that we should have one place for the abbreviated, precise, and current documentation. If you insist on demanding high standard from the sphinx docs, then we would keep that elsewhere (a readme file?, comments inside the code? wiki?), but we need it as the base reference for all the other documentation.

willingc, 2015-11-16:
@deshipu I agree that the API documentation should be clear, concise, and readable. After the source code itself, the API documentation should be the definitive resource for commands, functions, and input/output. You are doing a good job on the API docs.

I do believe that the API docs can and should be simple, friendly, and comprehensive. I think a few steps would be valuable to make it easier to do that:

1. Make it a point to review and merge documentation PRs quickly (goal: less than 3 days). This keeps info current for all.
2. Reorganize the docs source directory (``conf.py`` and master ``index.py`` at top-level) with subdirectories for:
  * ``api`` (definitive API reference - technical but clear - essentially what @deshipu has been working on), 
  * ``basics`` (hw and sw micro:bit info that is specific to micropython with docs linking to outside resources as needed), 
  * ``userdocs`` (11 yr old and educator accessible), 
  * ``examples`` (tutorials, descriptions of included games, etc), and 
  * ``project`` (contributions, workflow, filing issues).
3. Automate doc builds using tox, travis, etc. to check links, clean builds, etc.

While this may seem like it would be more time intensive, I assure you it will not be. It simply provides for more atomic management of documentation to better match the purposes of necessary documentation.

ntoll, 2015-11-16:
What @willingc said.

@deshipu ... I'm an engineer, writer and teacher and I spend a lot of time and effort on this project. I can't claim excellence but there are more of us in this skill space in the Venn diagram than you may think. ;-)

I'm working out how we can do a "write the docs" type process around this. http://www.writethedocs.org/

willingc, 2015-11-16:
FYI. Though this article is a few years old, it does a good job of capturing the layers of documentation http://blog.parse.com/learn/engineering/designing-great-api-docs/

For context, the docs that @deshipu has been working on would be along the lines of "API Reference".

ntoll, 2015-11-30:
See http://microbit-micropython.readthedocs.org/en/latest/

Amazing work @deshipu et al..! You win the internet! :-)

closed 2015-11-30 by ntoll

willingc, 2015-11-30:
@deshipu API docs :heart: and thank you!