bbcmicrobit/micropython #77: Create a built in tutorial

ntoll, 2015-11-07:
TL;DR the world's simplest tutorial and perhaps a good place to start for teachers. These should be as simple as possible. For example:

```
>>> import tutorial
Welcome to the built in tutorial!

There are 6 lessons:

0 - How to use MicroPython
1 - Using the display
2 - Buttons
3 - Moving the micr:obit
4 - Input/Output
5 - Advanced stuff

tutorial.show(0) displays the first lesson.
tutorial.list() displays this message again.
>>> tutorial(0)
How to use MicroPython

... some text no wider than 79 characters or longer than 20 lines ...
```

There should be a secret lesson 6 (top secret cool stuff) that is only referenced at the end of lesson 5. ;-)

- [ ] How to use MicroPython - a simple introduction to the REPL, Python and the micro:bit.
- [ ] Using the display - scrolling text and showing images.
- [ ] Buttons - waiting for input and reacting as a result.
- [ ] Moving the micro:bit - using the compass and accelerometer
- [ ] Input/Output - plugging things in via the pins
- [ ] Advanced stuff - animations and making sounds
- [ ] Top secret cool stuff - all about the Easter eggs and where to find out more

souravsingh, 2015-11-19:
I would like to work on the issue.

souravsingh, 2015-11-26:
How do I start on the issue?

whaleygeek, 2015-11-26:
I presume you're not talking about this being built into the standard build image? Otherwise it looks like you're gonna burn a lot of flash space with all that text. Please save some flash memory for a persistent data store to allow data-logging applications. Or consider compressing the text, at least.

Do you think teachers will use a command line tutorial, when a you tube video series would be much more useful perhaps (and Nicholas has made a load of them already!)? Or if you want to do a text tutorial, why not have a scriptid called "python tutorial" they can just download to their micro:bit, even a tutorial button in the editor that downloads the tutorial hex file?

souravsingh, 2015-11-28:
I believe creating inbuilt tutorial will be a good idea.Would love to know what @ntoll thinks about this.

souravsingh, 2015-12-04:
I am working on the issue right now.

ntoll, 2015-12-04:
HI @souravsingh - the first thing we should do is get the basic mechanism of the tutorial figured out. The initial idea given above should be refined. For example, we shouldn't use the `show` method since it's already used in a different context elsewhere (the display namespace). How about a `learn` method instead: `tutorial.learn(1)`..?

The actual content of the tutorial module can be worked out later... Each of the "lessons" should only contain "lesson 1" (for example) just for illustrative purposes.

Sorry for not getting back sooner... work is pretty busy at the moment.

souravsingh, 2015-12-04:
Alright @ntoll,Thanks for the guidance

carlosperate, 2015-12-04:
> How about a learn method instead: tutorial.learn(1)..?

>The actual content of the tutorial module can be worked out later... Each of the "lessons" should only contain "lesson 1" (for example) just for illustrative purposes.

`tutorial.lesson(1)` sounds quite good to me.

ntoll, 2015-12-04:
+1 `tutorial.lesson(i)`