bbcmicrobit/micropython #88: Investigate the situation of pull-up resistors for the I²C and SPI protocols

deshipu, 2015-11-09:
We don't yet know if there will be external pull-up resistors required for using the I²C and SPI protocols.
Once that information becomes available, we should update the documentation to include it.

whaleygeek, 2015-11-12:
I2C drivers are specified as open drain, which means they always need pullup resistors. This is so that the dominant and recessive bus states can be implemented (and thus the ACK/NAK bit and the more advanced multi master modes supported). Pullup resistors are traditionally external, because it depends how long and how fast the bus is run (bus capacitance) as to which values are chosen, thus it is dependent on how the system is built, not how the master (in this case the micro:bit) is built.

SPI has a definite master and slave, and does not require pullup resistors in the same way as I2C, as the protocol drives the bus hard high or hard low. For long SPI busses I've sometimes needed termination resistors to prevent reflections.

deshipu, 2015-11-12:
They may be traditionally external, but if you look at the Arduino Wire library, you will note that the I²C master sets the pins to INPUT_PULLUP.  I think this is done to make it easier to wire small projects, and it would also make sense for micro:bit (you can always add more pullups in parallel, to lower the resistance if your wires are long), so I'm not sure.

ntoll, 2015-11-12:
I guess @jaustin from ARM will be able to shed light on this. It was, after all, his team that designed the board.

Jonny?

msprotz, 2015-11-12:
There are internal pullups for the i2c sda and sdl ports. For one of my projects I added some 2.2k standard resistors myself for good measure, but technically they aren't needed.

deshipu, 2015-11-12:
Thank you everyone for the input, I added a section on connecting to the I²C docs.

carlosperate, 2015-11-12:
Does the current API allow selection of the internal pull up/downs? I didn't see anything in the current pin documentation pull request.

dpgeorge, 2015-11-13:
> Does the current API allow selection of the internal pull up/downs?

No, there is currently no way to explicitly configure a pin/gpio.  It's all done automagically when you use the pin in a specific way (eg call pin.digital_read).  Remember this is made for 11yo kids to use!

carlosperate, 2015-11-13:
Well, to follow the same convention and if we want to activate the internal pulls, then I2c read and write functions should save the current GPIO configuration state, activate the pull up, and then restore it back to its original saved stated. This could leave the lines floating and possibly causes issues, so I guess it needs to be decided if there is any merit in doing this, after all if the users are interfacing with an I2C device, they are most likely breadboarding already and adding physical resistors would not only be simple, but also part of the learning experience of using this type of protocol.

(As a side note, I had a look at the pin code, and I assume that the implementation of methods like `setDigitalValue` are probably part of the DAL still not released, so while it looks like current pin configuration state is not saved, that functionality might be hidden in that lower layer).

commit e276f6d8 (2015-11-24): Clarify connecting in I2C docs

Fixes #88
closed 2015-11-24 by dpgeorge