bbcmicrobit/micropython #91: microbit.display.print() easy to confuse with build-in print()

deshipu, 2015-11-11:
Maybe I'm paranoid, but we are setting the kids up for mistakes here.

First of all, we are introducing a function that is named the same as a build-in global function in Python. That means that whenever a teacher or a book mentions "print", there will be some confusion as to which print is meant, especially in spoken communication.

Second, we have two functions with the same name, but completely different parameters and functions. Python's print only accepts strings, and not images, and doesn't have a "delay" parameter. The display.print draws images and doesn't accept any of standard Python parameters for print. This seems like a small deal, but if you litter the language and libraries with such small inconsistencies, you get PHP -- that is, a language that takes considerable effort to learn and use properly, but is easy to use improperly.

I'm not sure what can actually be done at this point, it may be too late.

markshannon, 2015-11-11:
Touch develop for the microbit has `plot image`, `show string`, `show number` and `show animation`.
Ignoring the obvious odd one out in the above list, `display.show` would be better in terms of consistency with touch develop.
I would be happy to change it.
`microbit.display.show()` or `microbit.show`?

I don't it is too late, just yet, but we do need to finalise the exact API quite soon.

ntoll, 2015-11-11:
I agree we should change it for the reasons @deshipu explains. I quite like ``display.draw``.

@markshannon so ``microbit`` is automatically imported. It feels natural (to me anyway) to use ``display.draw`` since that's scoped nicely and reads like "get the display to draw...".

Thoughts?

deshipu, 2015-11-11:
I think either ``display.draw`` and ``display.show`` would work. I suppose ``display.show("some text")`` looks a little better than ``display.draw("some text")``, to my non-native-speaker eyes at least.

ntoll, 2015-11-11:
+1 on ``show`` because @deshipu's eyeball mk1 is correct. ;-)

carlosperate, 2015-11-11:
I would also agree that `show` seems pretty straight forward and works for both strings and images. On top of that is the same keyword that Microsoft's touchdevelop and blockly use.

dpgeorge, 2015-11-13:
I'm ok with changing `display.print()` to `display.show()`.  [Reason for the print name is because that's how the DAL calls it; they have print, scroll and animate.]

dpgeorge, 2015-11-17:
display.print is renamed to display.show in 536043811e9d41cae3f35bbf442c9ee856c2dc8d.

closed 2015-11-17 by dpgeorge