bbcmicrobit/micropython #92: Make series of build-in images easier to use

deshipu, 2015-11-11:
Right now we have two series of images in ``microbit.Image``, clock faces and arrows, and they all have names like ``CLOCK11``, which makes it awkward to access them automatically based on a value of a varibale, or to iterate over them.

It would be nice to have some way of refering to those collections by the collection name and the number in sequence, or, in case of arrows, maybe even name of of the direction as a string.

Perhaps it would be also nice if we could refer to all the other build-in images through a string with their name.

ntoll, 2015-11-12:
Given discussion elsewhere, how about something like...

```
Image.CLOCKFACE = (Image.CLOCK12, Image.CLOCK1...., Image.CLOCK11)
```

and 

```
Image.ARROWS = (Image.ARROW_N, Image.ARROW_NE..., Image.ARROW_NW)
```

These would be relatively simple to implement. Ergo, I'll add to milestone 0.9. ;-)

dpgeorge, 2015-11-13:
Making CLOCKFACE/ARROWS/etc a list of images might be confusing because then you have some constants which are images and some which are lists and it's not clear which are which just by the name.  Maybe prefix the name by `ALL_`, ie `ALL_CLOCKS` and `ALL_ARROWS`?

> Perhaps it would be also nice if we could refer to all the other build-in images through a string with their name.

How about: `getattr(Images, "CLOCK1")`?

deshipu, 2015-11-13:
It's already confusing, because you also have some that are methods of the Image class and some that are python internal stuff in there, and there is no way to just list the ones that are images.

> How about: ``getattr(Image, "CLOCK1")``?
How about ``exec("image = Image.{}".format("CLOCK1"))"``? Because that is what you will see used in this situation.

markshannon, 2015-11-24:
I think we may need less confusing name than `CLOCKFACE` and `ARROWS`.
So what are those names? I don't know. Anyone have any good suggestions?

deshipu, 2015-11-24:
How about removing the images from inside the ``Image`` class and putting them in separate collections?
Then there could be ``IMAGE.HEART``, ``ARROW[2]`` or ``ARROW.NE``, and ``CLOCK[12]``.

Other ideas:

``ARROW_COLLECTION``, ``ARROW_SEQUENCE``, ``ARROW_SEQ``, ``ARROW_FRAMES``, ``ARROW_IMAGES``, ``ROTATING_ARROW`` (it can be passed to ``display.show`` as is), ``SPINNING_ARROW``

willingc, 2015-11-25:
@markshannon Some other suggestions:
 ``FACE`` or ``DIAL``; 
``HANDS`` or ``BIG_HAND``, ``LITTLE_HAND``, ``HOUR_HAND``, ``MINUTE_HAND``;  ``TIME_12``...``TIME_6``...``TIME_11`` or ``HOUR_12``...``HOUR_6``...``HOUR_11``;
``NORTH`` ``SOUTH`` ``EAST`` ``WEST`` ``NE`` ``NW`` ``SE`` ``SW`` (shortening to N, S, E, W) as needed.

markshannon, 2015-11-25:
I think @dpgeorge's point was that names need to make it clear (just from the names) that the object is a collection of images, not a single image.
 How about `ALL_CLOCKS` and `ALL_ARROWS`?

deshipu, 2015-11-25:
> How about ALL_CLOCKS and ALL_ARROWS?

Love it!

dpgeorge, 2015-11-25:
+1 to ALL_CLOCKS and ALL_ARROWS.

deshipu, 2015-11-25:
Or we could do ``MUCH_ARROW`` and ``MANY_TIME`` and then add ``Image.DOGE`` ;)

carlosperate, 2015-11-25:
Just as a general though, would it be more convenient to have the "collection identification" as suffix rather and a prefix? that way with autocompletion a user could see all related options without having to know that the collection names starts with "ALL" or any other prefix.

deshipu, 2015-11-25:
And with a prefix they can easily see all collections. No idea which is more useful.

We don't have a suffix that would be short and wouldn't be confusing (can't use LIST, because it's a tuple, TUPLE is not commonly understood, SEQUENCE is a mouthfull, etc.), and I'm not sure it's actually worth dwelling on this too much. Let's pick an option that is not too horrible and roll with it.

markshannon, 2015-11-29:
Given the lack of further comments, I'm going with ALL_CLOCKS and ALL_ARROWS.

ntoll, 2015-11-30:
Fixed by #102

closed 2015-11-30 by ntoll