bbcmicrobit/micropython #93: Image string micro-language is hard to read

deshipu, 2015-11-11:
Right now, when we create a new image by passing in a string, we use rows of digits, separated by newlines, and whitespace is treated the same as digit 0. This leads to code that looks like this:

    image = Image("9   9\n 9 9 \n  9  \n 9 9 \n9   9")
    image = Image("""\
    9   9
     9 9
      9
     9 9
    9   9""")

If we modified that micro-language a little, we could improve the situation. First, ignore empty lines an whitespace:

    image = Image("""
        90009
        09090
        00900
        09090
        90009
    """)

Second, treat ":" (or some miliar punctuation character) as newline:

    image = Image("90009:09090:00900:09090:90009")

ntoll, 2015-11-12:
@markshannon what do you think..? I like the idea of making this a lot easier for kids to read in terms of both creating and looking at our image implementation. Given this is an API change, I'll add this to 0.9 milestone.

markshannon, 2015-11-12:
It is easy enough to implement any or all of the above suggestions.
I'll defer to others as to what is the most usable.

dpgeorge, 2015-11-13:
I do prefer "01234:56789:00000:01234:56789".

I think there should be one way to do it, not multiple options.

I don't think we need to support ignoring of whitespace.  You can just use implicit string concat:
```python
img = Image(
    "12345:"
    "00000:"
    "99999:"
)
```

carlosperate, 2015-11-13:
I agree with @dpgeorge, whatever the implementation, is probably better to stick the the "There's Only One Way To Do It" mantra, and then use string concatenation (with or without a character delimiter) to represent it in a more user friendly format

ntoll, 2015-11-13:
+1 with the added caveat that this one true way is hammered home in the documentation (along with explaining how to do string concat).

ntoll, 2015-11-18:
OK... I guess we have a "winner" in @dpgeorge's suggestion.

@markshannon care to do the honours?

markshannon, 2015-11-18:
Sure. I plan to fix various things for 0.9 over the weekend.

markshannon, 2015-11-29:
Keeping this issue open as a reminder to delete `\n` as a terminator, only allowing colons.

ntoll, 2015-11-30:
Fixed by #103

closed 2015-11-30 by ntoll

carlosperate, 2015-11-30:
Does this mean the line break terminators will definitely stay?