bbcmicrobit/micropython #95: Kalman filter

deshipu, 2015-11-16:
The micro:bit has both a compass and an accelerometer build-in, and they can be both used to asses the orientation of the board in space. However, the output of those devices is usually pretty noisy and difficult to use directly.

I wonder if it would make sense to include an implementation of the Kalman filter (https://en.wikipedia.org/wiki/Kalman_filter) in there, so that the two noisy inputs could be integrated into a more useful signal.

Sure, this could be done by the user, however, Kalman filter is quite hard to understand (especially for 11 year old children) and difficult to tune (there are a couple of papers on the subject, but they all admit using trial-and-error in the end). Including an already implemented and tuned filter could be a great help, and could increase the usefulness of those inputs greatly.

Neon22, 2015-11-17:
Is a GPL v2 code suitable for inclusion ? (written in Python)

dpgeorge, 2015-11-17:
> Is a GPL v2 code suitable for inclusion ? (written in Python)

Would be nice to keep away from GPL.  The reason being that it's a complicated license and we don't want to burden teachers/children with understanding licenses, they should just be able to code!

street-grease-coder, 2020-10-30:
To add some importance to this after 5 years... Kalman filters are incredibly useful and it would be extremely nice to have a simple library ready - this seems exceedingly hard to find

closed 2021-04-02 by deshipu