bbcmicrobit/micropython #96: Math module not available

deshipu, 2015-11-17:
I know that the microbit is limited in memory and excluding the math module probably lets shave off quite a bit of it, but I wonder if it's not too much. In particular, I think the basic trigonometry functions (sin, cos, tan, asin, acos, atan, atan2) and the square root function (sqrt) would be particularly useful. Yes, I know you can use the pow() or ** operator with fractional arguments, but still...

Is there any chance to have at least a part of the math module enabled?

ntoll, 2015-11-17:
From an educational perspective, this is probably a good thing to be able to do.

deshipu, 2015-11-17:
Also the log function.

dpgeorge, 2015-11-17:
Adding basic math functions mentioned above costs about 11k of code space (firmware jumps from roughly 175k to 186k).  Is it worth it?

ntoll, 2015-11-17:
I think it worth it for the following educational reasons:

* Kids will be learning about such simple functions in maths lessons and so it makes our solution compelling for maths teachers.
* I've already been contacted by someone at Edinburgh about teaching maths with Python on the micro:bit (so there's evidence that people want this).
* "Power" users will want to use such functions to make cool things.

dpgeorge, 2015-11-17:
Math module enabled in 3c08961bc94ae0905c4bee5cf808d61b13740f1b.  Costs 11924 bytes.

closed 2015-11-17 by dpgeorge