bbcmicrobit/micropython #104: Rationalise the display API.

markshannon, 2015-11-22:
Currently, to show an image, series of images or a string we have the following:
```python
display.show(string, delay=400) # prints the string to the display one character at a time
display.show(image, delay=400) # shows the image on the screen.
display.scroll(string, delay=400) # scrolls a string across the display (more exciting than display.print)
display.animate(image, delay, stride, start, wait=True, loop=False)
display.animate(iterable, delay, wait=True, loop=False)
```
There is some overlap between these and it is not always obvious which to use. The current interface is also inconsistent. `display.show()` doesn't take keyword arguments, but `display.animate` does, even though both take `delay`. `display.show("X", 500)` shows "X" forever. `display.show("XX", 500)` shows "X" for a second.

I propose to the following API:
Keep only `display.show()` and `display.scroll()`.
```python
display.show(image, delay=0, wait=True, loop=False, clear=False)
display.show(iterable, delay=400, wait=True, loop=False, clear=False)
display.scroll(string, delay=400)
```
The new version of `display.show()` should operate as `animate` does now and would take an iterable of images and/or single characters and display them sequentially. The single image variant would differ only in the default value for `delay`. Since strings are already sequences of single characters, we don't need to treat them specially.

`display.show` will take all the optional arguments that `animate` does now. The compulsory `delay` argument of `animate` would become optional and a sensible default provided. `display.show` will also gain a new optional parameter `clear` which would default to `False` and determine whether the display should be cleared at the end of the iteration.

`display.scroll()` would continue to work as it does now, creating a special iterator that is passed to `display.show()`: `display.scroll = lambda txt: display.show(scroller(txt))`.

This API change would entail some loss of functionality as `display.animate(image, delay, stride, start, ...)` will be gone. I don't see this as a problem as it can be easily reimplemented using `display.show` and `Image.crop`. In fact, it might make a nice example to do just that.

sparkslabs, 2015-11-22:
In case useful, +1 on the change :-)

(incidentally I've always thought of print as an internal thing, and
scrolling IMO is easier to read.)

Also, my understanding is this: since display.show when used with  a string
already displays a sequence of images, and you're building on that ? ie
display.show will show whatever is passed in. If it's passed in a string,
it shows one character at a time, and if what's passed in is an iterable
that returns images, it will show the images.

That strikes me as quite neat and tidy - since it allows something
logically like this:

    display.show("Hello")  # Not sure you're allowing this :-)

And

    display.show([emoji.HEART, emoji.FAV, emoji.SMILEY])

Would work the same way, which seems sense consistent.

That said, if I've misunderstood, it'd be useful to see a hint of an
example :)

Michael.

(Also, obviously in the example above, emoji.FAV would randomly become
emoji.HEART without any warning ;-) )

whaleygeek, 2015-11-26:
I was demonstrating this at a teacher event this week, and the display.animate *seemed* to have a mandatory delay parameter? All the other languages have a mode where the default delay is 400. 

Would anyone care to think through default parameters across the whole of this API (perhaps it needs another issue logged) - I was thinking of the display API only at the moment as that is what I experienced as a little 'bump' in the road as I was demonstrating it. 

Python has default parameters (unlike all of the other languages used with the exception of C++ but I think it will be a while before the kids try that!) it would be nice to think that sensible defaults were provided consistently across the whole API.

ntoll, 2015-11-26:
@whaleygeek an excellent point (we do "sane defaults" in the music API).

It should be relatively easy to do this. Given others use the default delay of 400 I see no reason why we shouldn't.

markshannon, 2015-11-27:
I was proposing that there be a default for all arguments (except the image(s)). So I'll take as a +1.
@whaleygeek Any thoughts about the rest of the proposed API?

markshannon, 2015-11-29:
I would like to get the API finalised. 
Implementing won't be difficult as it is little more than just renaming things, but I do want a consensus first.

ntoll, 2015-11-29:
@markshannon I like your API. It's a +1 from me.

deshipu, 2015-11-29:
I was going to suggest exactly the same think (maybe I even did in one of the comments on the docs, but it got lost), but since I wouldn't be able to implement it, I thought twice (I hate to tell others what they should do). I'm happy that someone wants to do it.

markshannon, 2015-11-29:
On further reflection, I think that we can drop both the `loop` and `clear` arguments.
If you want a loop, then 
```python
while True:
     display.show(...)
```
shouldn't be too hard.
Likewise, explicitly clearing
```
display.show(...)
display.clear()
```
is probably better.
For ultimate consistency `display.scroll` should take a `wait` parameter.
The new proposed API is:
```python
display.show(image, delay=0, wait=True)
display.show(iterable, delay=400, wait=True)
display.scroll(string, delay=400, wait=True)
```

markshannon, 2015-11-29:
Continuing my monologue...
I've had another rethink, and we do want a `clear` parameter and here is why:

`display.show(Image("X")` and `display.show("X")` should behave identically. 
But `"X"` is an iterable, and `Image("X")` is not. 
Therefore the default delay for both images and iterables must be the same (or they aren't behaving identically). 
Not only that, but `display.show(Image("X")` and `display.show("X")` should return immediately and not pause for `delay` milliseconds. 
The way to achieve this is to treat a single image as a sequence of one and delay *in between* images, but not after the last image.
However, that would cause other problems as it means that the code:
```python
display.show(Image.ALL_CLOCKS)
display.clear()
```
Only displays the final hand fleetingly.
By adding a `clear` parameter which effectively adds a blank image to the end of the sequences of images,
we get the behaviour we want.
In summary:
`display.show(img)` where `img` is an image behaves *exactly* like `display.show([img])` and
`display.show(seq, clear=True)` behaves exactly as
`display.show(follows(seq, Image.BLANK))` where 
```python
def follows(seq, item):
     yield from seq
     yield item
```
This API seems to do what you would expect:
`display.show("X", delay=1000, clear=True)`   displays "X" for 1 second then clears the display
`display.show(Image("X"), delay=1000, clear=True)` does exactly the same.
(Note that both the above are equivalent to `display.show([Image("X"), Image()], delay=1000)`)
`display.show((Image.SAD, Image.HAPPY))`, displays SAD for 400ms, then HAPPY for ever.
`display.show((Image.SAD, Image.HAPPY), clear=True)` displays SAD for 400ms, then HAPPY for 400ms
`display.show(Image.ALL_CLOCKS, clear=True)` displays each clock in turn for 400ms.

The scrolling iterator backing `display.scroll` produces a blank image as its last image, so will continue to function as before.

dpgeorge, 2015-11-29:
If you want an "animation" to run continuously in the background then you need `loop=True` and `wait=False`.  This mirrors the music module.

markshannon, 2015-12-03:
There are several issues with adding a `loop` parameter that lead to me to decide that it should be dropped.
In no particular order:

* Animation is not really a background thing in the way that sound can be. The animation is likely to be the focus of attention.
* Loops ought to be TOWTDI for repeated operations. In general we are trying to avoid a callback style.
* What happens for iterators? Do we record the output of the iterator? Repeating an exhausted iterator is not going to work.
* If you want an infinitely repeating sequence, then it is not that hard to create one. I'm not saying that is easy for an 11 year old, but we could provide a recipe in the documentation, and maybe implement `itertools.cycle`.
* The more parameters a method has, the harder it is to learn. I was reluctant to add `clear` but I could find no way to achieve a consistent API otherwise.

dpgeorge, 2015-12-04:
But loop is already implemented for scroll and animate.  For iterators it saves the iterable and calls mp_getiter on each repeated loop.

markshannon, 2016-01-16:
Time to revisit this before 1.0.
I was reluctant to restore `loop` to the API. The interaction between iterators and the `loop` parameter is messy, but I've added it back in. See https://github.com/bbcmicrobit/micropython/issues/153 for explanation.
Which brings me back to the API that I original proposed :smile:

closed 2016-01-21 by markshannon