bbcmicrobit/micropython #105: Inconsistent behavior of PWM

deshipu, 2015-11-24:
The PWM output from microbit behaves really strangely. I recorded a video to show it:
https://www.youtube.com/watch?v=bG1xZdq2iX8

(The servo is a continuous-rotation servo, which rotates one way when the signal is below 1500µs, and the other way when it's above.)

I don't have a logic analyzer or an osciloscope to actually see what is happening with the signal, but I'm quite sure this is not right.

Right after setting the PWM period to 20 (which should correspond to 50Hz, which is what the servo expects) and doing ``write_analog(120)``, the servo starts to rotate, however, the direction is wrong. Repeating write_analog() has no effect. Now, if I set the period again, to the same value, and call ``write_analog(120)`` again, the servo stops, does several hiccups and starts rotating correct way. Repeated ``write_analog(120)`` sometimes result in the servo reversing its direction...

dpgeorge, 2015-11-24:
Probably this is an issue for the DAL, since we are a thin wrapper over it.  But I guess first we should formulate a precise description of the problem.

deshipu, 2015-11-24:
Today evening I will go to the local hackerspace and see if they can let me use their osciloscope.

deshipu, 2015-11-24:
Also, I can't report an issue with the dal, because that repository is hidden.

dpgeorge, 2015-11-24:
You'll notice that the music output is a bit "funny" as well, because the PWM is not very clean and jitters somewhat.  That could be related.

deshipu, 2015-11-24:
OK, so after playing with the analog scope a little bit (awesome device!), I'm a little bit wiser about what is happening. There are actually two separate problems here.

1. This servo is not designed to work with 3V. I assumed it will work, because it works fine in my other robots that are powered by 3.7V LiPo, but apparently a set of two 1.5V alkaline batteries is not enough. The random direction changes were caused by this -- the servo took random noise at the top of the square wave as the signal, and thus it acted as if the duty cycle was very short. Powering the robot from a LiPo fixes that.

2. After setting the PWM period for the first time, it somehow gets set to a much higher frequency. I don't know the scope enough to actually read what the frequency was, but it was an order of magnitude higher. Only after doing some ``write_analog`` and setting the frequency again, it got set to a right value.

I will upload some photos from the testing, maybe someone will see something more from them.

deshipu, 2015-11-25:
I reported it for DAL: https://github.com/lancaster-university/microbit-dal/issues/51

whaleygeek, 2015-11-26:
I see this in the other languages all the time too. Sometimes it's related to the order that you call the pitch/value commands, which suggests to me it's a register setup hazard in the DAL, and/or a period calculation overflow or loss of precision in the integer maths associated with it.

deshipu, 2016-01-14:
The issue has been fixed in DAL in the master branch. It should be tagged soon, then we can change the version in ``moudules.json`` and close this ticket.

dpgeorge, 2016-01-22:
DAL has been updated to v1.4.6 which includes PWM fixes.  So closing this issue.

closed 2016-01-22 by dpgeorge