bbcmicrobit/micropython #107: Mangled REPL text when copy-pasting

deshipu, 2015-11-24:
When copy-pasting commands into the REPL over the USB serial, I get the characters echoed back in wrong order. For instance, copy-pasting those three lines, a line at a time:

    import microbit
    microbit.pin0.set_analog_period(20)
    microbit.pin0.write_analog(20)

results in this output:

    >>> ootort microbit
    >>> rricro0bit.pin0.set_analog_period(20)
    >>> rric_l2it.pin0.write_analog(20)

dpgeorge, 2015-11-24:
Yes, this has been an issue since the start (I use picocom).  The code still runs, so it's getting in correctly, but not back to the PC in the right order.

There's also an issue if you paste too much text that it starts to return 2x each character.

There are many layers in the chain:

1. USB CDC driver on host PC.
2. USB CDC device firmware on the microbit in the Freescale support chip, converting to UART.
3. nRF51 UART hardware.
4. mbed UART driver.
5. MicroPython.

I suspect it's 2, since all other layers are well tested in other settings.

dpgeorge, 2015-11-29:
I contacted ARM about this and it seems that it's number 4 above (mbed UART driver).  They are going to fix it (hopefully!).

deshipu, 2015-11-30:
I just noticed another curious thing about the REPL. After I press ctrl+L, all the characters are echoed twice...

dpgeorge, 2015-11-30:
> I just noticed another curious thing about the REPL. After I press ctrl+L, all the characters are echoed twice...

I cannot reproduce this (using picocom).  But I have seen the behaviour before.

Neon22, 2015-11-30:
I had this behaviour when I pasted a large track of code hoping it
    was a way to get it onto the Microbit.
    The terminal then started echoing. It cleared only when reset.
    I assumed the echo had somehow been triggered by the board receiving
    a series of bytes setting it into echo mode.
    
    I found the best way to get code on was the method using: 
        tools/pyboard.py ttyACM0 yourcode.py
        Which also opened the REPL so I could see prints etc.
    Needless(?) to say having a gtkterm open at the same time was a
    flawed way of thinking about it.
    
    
    On 12/1/2015 8:21 AM, Damien George
      wrote:
    
    
      
        I just noticed another curious thing about the REPL. After I
          press ctrl+L, all the characters are echoed twice...
      
      I cannot reproduce this (using picocom). But I have seen the
        behaviour before.

dpgeorge, 2018-03-20:
@microbit-carlos I can confirm that the latest v1 code does seem to resolve this issue.  I didn't do anything explicitly to fix this so I suspect something in the dependent components has changed/been fixed.  Indeed it looks like the version of mbed-classic that is now used by the DAL is a patched version of upstream mbed, see https://github.com/lancaster-university/mbed-classic.  In particular this patch made changes to internal timing which very likely fixed the issue: https://github.com/lancaster-university/mbed-classic/commit/6888edef4c8ba0e1a0597eabeceb60a779e9ce2b .  These changes were made in Feb 2016 which was after this bug was reported so it does make sense that that custom version of mbed-classic has fixed this issue.

It's now possible to paste up to about 400 bytes of text at once into the REPL without any issue.  Bytes beyond that sometimes get lost but that is most like because the internal UART buffer is overflowing and can't handle so much incoming text at once (this happens on other MicroPython ports).  But crucially, pasting too many bytes no longer puts the UART into a funny mode where it returns 2x the characters.

Given the above, let's close this issue because it looks to be properly fixed.

closed 2018-03-20 by dpgeorge