bbcmicrobit/micropython #109: cstddef: No such file or directory

deshipu, 2015-11-24:
I'm trying to compile the latest master on Ubuntu, following the instructions from README, and I stumbled upon this problem:

```
(micropython)sheep@ghostwheel:~/x/micropython$ yt target bbc-microbit-classic-gcc-nosd
(micropython)sheep@ghostwheel:~/x/micropython$ yt up
info: get versions for bbc-microbit-classic-gcc-nosd
info: get versions for mbed-gcc
info: get versions for mbed-classic
info: get versions for ble
warning: Using head of "master" branch for "ble", not a tagged version
info: update outdated: ble@0.4.3 -> ble@master from GitHub lancaster-university/BLE_API
info: download ble@master from GitHub lancaster-university/BLE_API
info: get versions for ble-nrf51822
warning: Using head of "master" branch for "ble-nrf51822", not a tagged version
info: update outdated: ble-nrf51822@0.4.3 -> ble-nrf51822@master from GitHub lancaster-university/nRF51822
info: download ble-nrf51822@master from GitHub lancaster-university/nRF51822
info: get versions for microbit-dal
warning: Using head of "master" branch for "microbit-dal", not a tagged version
info: update outdated: microbit-dal@1.3.8 -> microbit-dal@master from GitHub lancaster-university/microbit-dal
info: download microbit-dal@master from GitHub lancaster-university/microbit-dal
(micropython)sheep@ghostwheel:~/x/micropython$ yt build
info: generate for target: bbc-microbit-classic-gcc-nosd 0.1.2 at /home/sheep/x/micropython/yotta_targets/bbc-microbit-classic-gcc-nosd
GCC version is: 4.9.3
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sheep/x/micropython/build/bbc-microbit-classic-gcc-nosd
[6/249] Building CXX object ym/mbed-cl...mbed-classic.dir/common/BusInOut.cpp.o
FAILED: /usr/bin/arm-none-eabi-g++   -fno-exceptions -fno-unwind-tables -ffunction-sections -fdata-sections -Wall -Wextra -fno-rtti -fno-threadsafe-statics -mcpu=cortex-m0 -mthumb -D__thumb2__ -Os -g -gdwarf-3 -DNDEBUG -Igenerated/include -I/home/sheep/x/micropython -I/home/sheep/x/micropython/yotta_modules/mbed-classic -I/home/sheep/x/micropython/yotta_modules/ble -I/home/sheep/x/micropython/yotta_modules/ble-nrf51822 -I/home/sheep/x/micropython/yotta_modules/microbit-dal -I/home/sheep/x/micropython/yotta_modules/mbed-classic/api -I/home/sheep/x/micropython/yotta_modules/mbed-classic/hal -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/cmsis -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822 -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/TARGET_NRF51_MICROBIT -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk/components -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk/components/libraries -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk/components/libraries/crc16 -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk/components/libraries/scheduler -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk/components/libraries/util -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/s130_nrf51822_1_0_0 -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/s110_nrf51822_8_0_0 -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/cmsis/TARGET_NORDIC -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/cmsis/TARGET_NORDIC/TARGET_MCU_NRF51822 -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/cmsis/TARGET_NORDIC/TARGET_MCU_NRF51822/TOOLCHAIN_GCC_ARM -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/cmsis/TARGET_NORDIC/TARGET_MCU_NRF51822/TOOLCHAIN_GCC_ARM/TARGET_MCU_NRF51_16K_S110    -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -DMBED_OPERATORS -DNRF51 -DTARGET_NORDIC -DTARGET_M0 -D__MBED__=1 -DMCU_NORDIC_16K -DTARGET_NRF51_MICROBIT -DTARGET_MCU_NORDIC_16K -DTARGET_MCU_NRF51_16K_S110  -DTARGET_NRF_LFCLK_RC -DTARGET_MCU_NORDIC_16K -D__CORTEX_M0 -DARM_MATH_CM0 -DNO_BLE -include "/home/sheep/x/micropython/build/bbc-microbit-classic-gcc-nosd/yotta_config.h" -w -MMD -MT ym/mbed-classic/existing/CMakeFiles/mbed-classic.dir/common/BusInOut.cpp.o -MF ym/mbed-classic/existing/CMakeFiles/mbed-classic.dir/common/BusInOut.cpp.o.d -o ym/mbed-classic/existing/CMakeFiles/mbed-classic.dir/common/BusInOut.cpp.o -c /home/sheep/x/micropython/yotta_modules/mbed-classic/common/BusInOut.cpp
In file included from /home/sheep/x/micropython/yotta_modules/mbed-classic/api/DigitalInOut.h:19:0,
                 from /home/sheep/x/micropython/yotta_modules/mbed-classic/api/BusInOut.h:19,
                 from /home/sheep/x/micropython/yotta_modules/mbed-classic/common/BusInOut.cpp:16:
/home/sheep/x/micropython/yotta_modules/mbed-classic/api/platform.h:25:19: fatal error: cstddef: No such file or directory
 #include <cstddef>
                   ^
compilation terminated.
[6/249] Building CXX object ym/mbed-cl...s/mbed-classic.dir/common/BusOut.cpp.o
FAILED: /usr/bin/arm-none-eabi-g++   -fno-exceptions -fno-unwind-tables -ffunction-sections -fdata-sections -Wall -Wextra -fno-rtti -fno-threadsafe-statics -mcpu=cortex-m0 -mthumb -D__thumb2__ -Os -g -gdwarf-3 -DNDEBUG -Igenerated/include -I/home/sheep/x/micropython -I/home/sheep/x/micropython/yotta_modules/mbed-classic -I/home/sheep/x/micropython/yotta_modules/ble -I/home/sheep/x/micropython/yotta_modules/ble-nrf51822 -I/home/sheep/x/micropython/yotta_modules/microbit-dal -I/home/sheep/x/micropython/yotta_modules/mbed-classic/api -I/home/sheep/x/micropython/yotta_modules/mbed-classic/hal -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/cmsis -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822 -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/TARGET_NRF51_MICROBIT -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk/components -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk/components/libraries -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk/components/libraries/crc16 -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk/components/libraries/scheduler -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk/components/libraries/util -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/s130_nrf51822_1_0_0 -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/s110_nrf51822_8_0_0 -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/cmsis/TARGET_NORDIC -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/cmsis/TARGET_NORDIC/TARGET_MCU_NRF51822 -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/cmsis/TARGET_NORDIC/TARGET_MCU_NRF51822/TOOLCHAIN_GCC_ARM -I/home/sheep/x/micropython/yotta_modules/mbed-classic/targets/cmsis/TARGET_NORDIC/TARGET_MCU_NRF51822/TOOLCHAIN_GCC_ARM/TARGET_MCU_NRF51_16K_S110    -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -DMBED_OPERATORS -DNRF51 -DTARGET_NORDIC -DTARGET_M0 -D__MBED__=1 -DMCU_NORDIC_16 [... cut ...] ropython/yotta_modules/mbed-classic/common/SPI.cpp
In file included from /home/sheep/x/micropython/yotta_modules/mbed-classic/api/SPI.h:19:0,
                 from /home/sheep/x/micropython/yotta_modules/mbed-classic/common/SPI.cpp:16:
/home/sheep/x/micropython/yotta_modules/mbed-classic/api/platform.h:25:19: fatal error: cstddef: No such file or directory
 #include <cstddef>
                   ^
compilation terminated.
ninja: build stopped: subcommand failed.
error: command ['ninja'] failed
```

dpgeorge, 2015-11-24:
Same problem reported here: https://mail.python.org/pipermail/microbit/2015-November/000602.html

deshipu, 2015-11-24:
Just to make sure, I have all the listed PPAs added, and I have both ``libstdc++-arm-none-eabi-newlib`` and ``libnewlib-arm-none-eabi`` installed.

dpgeorge, 2015-11-24:
Here is some possible clues/help: https://mail.python.org/pipermail/microbit/2015-November/000623.html

deshipu, 2015-11-24:
I fixed it with this command:

```
sudo ln -s /usr/lib/arm-none-eabi/include/c++/4.9 /usr/lib/arm-none-eabi/include/c++/4.9.3
```

closed 2015-11-24 by deshipu

carlosperate, 2016-04-15:
I've had this issue as well for Ubuntu 14.04, the only way I was able to fix it was to remove the default arm-none-eabi and force a specific version on apt-get install:

```
sudo apt-get remove binutils-arm-none-eabi gcc-arm-none-eabi
sudo apt-get install gcc-arm-none-eabi=4.9.3.2015q3-1trusty1
arm-none-eabi-gcc --version
```
You can find specific package versions for different Ubuntu releases on https://launchpad.net/~terry.guo/+archive/ubuntu/gcc-arm-embedded

JoeGlancy, 2016-04-18:
Off the top of my head (I do have a rather outdated environment setup guide
addressed this issue:
http://c0d3st0rm.xyz/microbit/micropython/setting-up-a-build-environment -
see step 4), I managed to fix this by adding the appropriate include
directory for the C++ headers in the CMakeLists.txt file for the
mbed-classic source.

carlosperate, 2016-04-18:
Wouldn't that mix the the use of an old compiler with newer C++ libraries? Even if that doesn't matter that much, and I might be wrong here, I think one small caveat would be finding the right include directory. It will depend on which distribution and which version you use as they can have the compilers configured differently (as mentioned in the email list, Ubuntu does that odd thing with the versioning, but even this behaviour is different pre or post 14.04).

JoeGlancy, 2016-04-18:
I *think* that those are the appropriate headers for the compiler
(`gcc-arm-none-eabi`), even so everything compiled fine. Also, I made two
build environments, one adding Terry's repo and one not -
`gcc-arm-none-eabi` installed and worked fine on both, so I assume that the
normal ARM compiler (which must be in the standard repos as it was a clean
install of 15.10) vs Terry's is the same? At any rate, I don't see any
issue (although I bet there are differences that I don't know about)
between compilers, or using the include directories mentioned above.

carlosperate, 2016-04-18:
>so I assume that the normal ARM compiler (which must be in the standard repos as it was a clean install of 15.10) vs Terry's is the same? 

Seems like, at least in trusty (14.04 LTS), their repos only have gcc-arm-none-eabi up to version 4.8.2, while terry's goes up to 4.9.x, and the newer proper ARM upstream ones I think are at around 5.3.x. So definitely different versions depending of where you get them from.

On trusty, without the newer 4.9.x compiler there are no additional headers in `/usr/include/newlib/`, and as far as I could find, they only seem to be in subdirectories of `/usr/lib/gcc/arm-none-eabi/` and ` /usr/lib/gcc/arm-none-eabi/`, which I would assume get automatically added to the path by the compiler itself? Either way, it is a lot easier to upgrade the compiler to get around this issue.

JoeGlancy, 2016-04-19:
Agreed. So why aren't we using ARM's upstream ones?

carlosperate, 2016-04-20:
Well, I'm not quite sure, this version seems to be working fine. Nothing against using the newer versions, but I'm not really aware of any pressing reasons to upgrade?

Also, there is this issue already opened: https://github.com/bbcmicrobit/micropython/issues/203

carlosperate, 2016-04-26:
Sounds like the DAL main devs are still using 4.9.x as well: https://github.com/lancaster-university/microbit-dal/issues/136

zydz, 2017-08-22:
Maybe you use ubuntu 14.04 and arm-none-eabi version is out of date. On my case I did the following steps and it works alright:
1 Please go download: 
https://launchpadlibrarian.net/287101520/gcc-arm-none-eabi-5_4-2016q3-20160926-linux.tar.bz2
2. uncompress it
3. export=/your/gcc-arm-none-eabi/path/bin:$PATH
4. rm -r /your/micropython-sourcecode/path/build/
5. yt build

1lzx, 2019-06-11:
/home/liu/micropython/yotta_modules/microbit-dal/source/drivers/MicroBitCompassCalibrator.cpp:229:31: error: in C++98 'centre' must be initialized by constructor, not by '{...}'
     Sample3D centre = { 0,0,0 };
                               ^
/home/liu/micropython/yotta_modules/microbit-dal/source/drivers/MicroBitCompassCalibrator.cpp:230:29: error: in C++98 'best' must be initialized by constructor, not by '{...}'
     Sample3D best = { 0,0,0 };
                             ^
[120/325] Building CXX object ym/microbit-dal/source/...eFiles/microbit-dal.dir/drivers/MicroBitCompass.cpp.o
ninja: build stopped: subcommand failed.

/**/
按照您的方法出现了以上的问题。