bbcmicrobit/micropython #113: install instructions - yotta

Neon22, 2015-11-28:
Following instructions in the readme.
I needed to change the line
```pip3 install yotta```
to
```sudo pip3 install yotta```
to get past permission to create directories under python3.4/dist-packages.
After change I was able to install yotta on ubuntu14.04LTS

deshipu, 2015-11-28:
Personally I don't like installing anything globally with pip like that, so I use a virtualenv. Then the pip command works without sudo.

deshipu, 2015-11-28:
In recent versions of pip, it also defaults to ``--user`` installs, which work in your home directory, not globally, and also should not require sudo.

Neon22, 2015-11-28:
well its as fresh as it comes - downloaded today and without "sudo pip3 ... " it did not create directories.
I have never used virtualenv - I can't comment on its workflow.

carlosperate, 2015-11-28:
The thing about virtualenvs is that if the user is already familiar with it, then they will be able to work it out themselves, the general instructions should be simple to follow.

willingc, 2015-11-30:
@Neon22 If you would like to try using virtualenv with Python3 on Ubuntu, I would recommend the installation instructions for the Django Girls workshop.

This section, http://tutorial.djangogirls.org/en/python_installation/index.html, explains how to install Python3.

This section, http://tutorial.djangogirls.org/en/django_installation/index.html, explains how to install and use virtualenv. Note: Ubuntu 14.04 specific steps.

Neon22, 2015-11-30:
After I have got the work done I will spend another day on it. took me a whole day to get the system runnig and the image doesn't work anyway.
I have a pdf of my process but can't attach it. I have quoted it below in case its useful
pretty sure Python3 was already on ubuntu1404LTS

'''
Getting a Linux machine setup to build micropython for the BBC microbit
Nov 2015

Initial machine setup
1.	Create new machine by installing linux14.04LTS from download and burn onto ISO disk, then booting machine from DVD. (ISO is >1MB so CD is no good)
2.	Alas system locks up on my old Dell Dimension C521 because Ubuntu desktop manager is too big for the graphic card's RAM. (scrambled screens and no kbd/mouse after a few seconds is the clue).
a.	Hunt around on web...   I need xfce4 - a lightweight desktop.
b.	Install xfce4 from another console. 
a.	sudo apt-get install xfce4
c.	Restart ubuntu and choose xfce instead of Gnome.
a.	When logging in - located in Session dropdown.
d.	All OK, memory is fine, system not locking up. We can now start working on the software.
Installing Git
3.	Need to install git - not included in Ubuntu14.04LTS distro.
a.	Follow instructions from web
i.	https://www.howtoforge.com/tutorial/install-git-and-github-on-ubuntu-14.04/
ii.	sudo apt-get install git
b.	Add your user info
i.	git config --global user.name "your user_name"
ii.	git config --global user.email "your email"
c.	You now have git installed and setup and you can clone from the web, or make new projects.

Compiling
4.	Instructions from micropython:
i.	https://github.com/micropython/micropython/wiki/Getting-Started
a.	Here we're just trying to get micropython environment setup for compiling. We are not targeting the microbit board yet.
b.	Followed directions for ARM ports on that page.
c.	Need to uninstall incorrect compiler first
i.	sudo apt-get remove binutils-arm-none-eabi gcc-arm-none-eabi
d.	Then add what we need to cross compile
ii.	sudo add-apt-repository ppa:terry.guo/gcc-arm-embedded
iii.	sudo apt-get update
iv.	sudo apt-get install gcc-arm-none-eabi
e.	Decided not to attempt to clone repo and compile the Micropython repo directly. It uses the same directory when cloning as the bbcmicro directory will use later on. Likely to cause me problems and I need to concentrate on the microbit side anyway
f.	You have now setup the environment to cross compile using the gcc compiler.
5.	Building the Microbit version
a.	Main difference is the use of Yotta. Which adds lots of dependencies and has the strangest(IMHO) mechanism which needs an active internet connection and Browser to compile.
b.	Instructions here:
i.	https://github.com/bbcmicrobit/micropython
c.	Getting Yotta setup is tricky as have to go to a further page for instructions and only use some of them.
i.	sudo add-apt-repository -y ppa:terry.guo/gcc-arm-embedded
ii.	sudo add-apt-repository -y ppa:pmiller-opensource/ppa
iii.	sudo apt-get update
iv.	sudo apt-get install cmake ninja-build gcc-arm-none-eabi srecord
d.	Get yotta in properly
i.	instructions from here: http://yottadocs.mbed.com/#installing
ii.	sudo apt-get install python-setuptools cmake build-essential ninja-build libffi-dev libssl-dev
iii.	sudo apt-get update
e.	Back to the bbc instructions but you need pip3 first:
i.	sudo apt-get install python3-pip
ii.	sudo pip3 install yotta
f.	Clone the bbc repo
i.	git clone https://github.com/bbcmicrobit/micropython
ii.	puts it in a directory called micropython probably located in your home directory
g.	You are now ready to compile
6.	Try running Yotta and generating a hex file
a.	to get inside the repo - cd micropython 
b.	yt target bbc-microbit-classic-gcc-nosd
i.	Suddenly it pops open a browser and we need to have an account on mbed.com 
ii.	so we signup and get an email to reply to, for registration
c.	So for it to complete, You'll have to do this step again and have the Browser open and be prompted to login again.
7.	Yotta next steps
a.	Fetch remote assets - yt up
i.	Will once again open a browser, demand login, and continue
b.	 Start the build - yt build
i.	Will not open a browser, and will compile to a build
c.	The build will be the .hex file in the directory:
i.	build/bbc-microbit-classic-gcc-nosd/source 
d.	You now have a compiled .hex file suitable for installing on the Microbit.
8.	Getting the hex file onto the Microbit.
a.	This turns out to be simple. You just copy the hex file onto the Microbit.
b.	If you connect it via USB, the Microbit will show up as a Disk drive on your desktop. 
c.	You can drop the .hex file into here.
d.	It will recognize the hex file and restart itself running that program.

'''

willingc, 2015-12-02:
Sections of message with steps done by @Neon22 should be included for documentation in #124. Can we close this issue since it will be addressed in #124? Initial PR #128 was merged yesterday.

Neon22, 2015-12-02:
My latest data is this
"""
Getting a Linux machine setup to build micropython for the BBC microbit
Nov 2015

Initial machine setup
1.	Create new machine by installing linux14.04LTS from download and burn onto ISO disk, then booting machine from DVD. (ISO is >1MB so CD is no good)
2.	Alas system locks up on my old Dell Dimension C521 because Ubuntu desktop manager is too big for the graphic card's RAM. (scrambled screens and no kbd/mouse after a few seconds is the clue).
a.	Hunt around on web...   I need xfce4 - a lightweight desktop.
b.	Install xfce4 from another console. 
a.	sudo apt-get install xfce4
c.	Restart ubuntu and choose xfce instead of Gnome.
a.	When logging in - located in Session dropdown.
d.	All OK, memory is fine, system not locking up. We can now start working on the software.
Installing Git
3.	Need to install git - not included in Ubuntu14.04LTS distro.
a.	Follow instructions from web
i.	https://www.howtoforge.com/tutorial/install-git-and-github-on-ubuntu-14.04/
ii.	sudo apt-get install git
b.	Add your user info
i.	git config --global user.name "your user_name"
ii.	git config --global user.email "your email"
c.	You now have git installed and setup and you can clone from the web, or make new projects.

Compiling
4.	Instructions from micropython:
i.	https://github.com/micropython/micropython/wiki/Getting-Started
a.	Here we're just trying to get micropython environment setup for compiling. We are not targeting the microbit board yet.
b.	Followed directions for ARM ports on that page.
c.	Need to uninstall incorrect compiler first
i.	sudo apt-get remove binutils-arm-none-eabi gcc-arm-none-eabi
d.	Then add what we need to cross compile
ii.	sudo add-apt-repository ppa:terry.guo/gcc-arm-embedded
iii.	sudo apt-get update
iv.	sudo apt-get install gcc-arm-none-eabi
e.	Decided not to attempt to clone repo and compile the Micropython repo directly. It uses the same directory when cloning as the bbcmicro directory will use later on. Likely to cause me problems and I need to concentrate on the microbit side anyway
f.	You have now setup the environment to cross compile using the gcc compiler.
5.	Building the Microbit version
a.	Main difference is the use of Yotta. Which adds lots of dependencies and has the strangest(IMHO) mechanism which needs an active internet connection and Browser to compile.
b.	Instructions here:
i.	https://github.com/bbcmicrobit/micropython
c.	Getting Yotta setup is tricky as have to go to a further page for instructions and only use some of them.
i.	sudo add-apt-repository -y ppa:terry.guo/gcc-arm-embedded
ii.	sudo add-apt-repository -y ppa:pmiller-opensource/ppa
iii.	sudo apt-get update
iv.	sudo apt-get install cmake ninja-build gcc-arm-none-eabi srecord
d.	Get yotta in properly
i.	instructions from here: http://yottadocs.mbed.com/#installing
ii.	sudo apt-get install python-setuptools cmake build-essential ninja-build libffi-dev libssl-dev
iii.	sudo apt-get update
e.	Back to the bbc instructions but you need pip3 first:
i.	sudo apt-get install python3-pip
ii.	sudo pip3 install yotta
f.	Clone the bbc repo
i.	git clone https://github.com/bbcmicrobit/micropython
ii.	puts it in a directory called micropython probably located in your home directory
g.	You are now ready to compile
6.	Try running Yotta and generating a hex file
a.	to get inside the repo - cd micropython 
b.	yt target bbc-microbit-classic-gcc-nosd
i.	Suddenly it pops open a browser and we need to have an account on mbed.com 
ii.	so we signup and get an email to reply to, for registration
c.	So for it to complete, You'll have to do this step again and have the Browser open and be prompted to login again.
7.	Yotta next steps
a.	Fetch remote assets - yt up
i.	Will once again open a browser, demand login, and continue
b.	 Start the build - yt build
i.	Will not open a browser, and will compile to a build
c.	The build will be the .hex file in the directory:
i.	build/bbc-microbit-classic-gcc-nosd/source 
[... 23 lines cut ...]
i.	use "sudo apt-get install gtkterm" to get it.
ii.	It has a configuration profile file you can save to. (defines port and speed settings and it will open using that each time).
iii.	The usb tty port will be discovered and presented to you in the Configuration menu automatically.
c.	Run it and navigate to menu Configuration/Set port. 
d.	Paste in the ttyACM0 you found above and also these settings:
i.	baud 115200, 8 bits, no parity, 1 Stop bit
e.	Save it as the default configuration so it uses these settings every time it opens.
5.	Initiate the connection and you should see the familiar python prompt.
6.	You are now ready to write Python code

Remote controlling the Microbit
1.	You can get a program onto the board and run it by calling the pyboard program in the Tools directory
a.	E.g. ./tools/pyboard.py /dev/ttyACM0 your_code.py
b.	This will open an interactive link to the board. You will see the output of print statements and error messages as the code runs.
c.	It is critical to use the reset button on the board every time you are about to upload a new version of your file. Otherwise you may receive misleading error messages and waste a lot of time trying to fix a problem that does not exist.
d.	Using this at the same time as having another serial terminal open is not going to work. You may receive conflicting and confusing output or your program may fail to load.
2.	You can also pack a python program up with the hex file so it runs whenever the Microbit starts up. Like this:
[... cut ...]
i.	E.g. tools/makecombinedhex.py build/bbc-microbit-classic-gcc-nosd/source/microbit-micropython.hex microbit-fire.py > send.hex
b.	When you do this there is no longer a REPL but the program will run as soon as power is applied.

Important side Notes:
If you login using the serial terminal and use the interactive REPL on the board you will see an implicit "import microbit as *" has already occurred. This may be confusing when you start to upload files as you'll need to add "import microbit" to the head of your py files.

"""

remay, 2015-12-03:
> When the Microbit is plugged into Ubuntu system it will detect that and do the work required
> to connect up the devices it sees.
> 1. One of these is the disk drive aspect
> a. which will show up on your [...]

I've been meaning to look further at mbed-ls (https://github.com/ARMmbed/mbed-ls) to see if it can really simplify those steps - I suspect it wouldn't need much work to only identify the micro:bit, and report the disk mount point and serial port, which could be used to simplify the process a bit.

> E.g. ./tools/pyboard.py /dev/ttyACM0 your_code.py
Could potentially become just `./tools/pyboard.py your_code.py`.

I have no time to look further any time soon, but if anyone else wanted to then I think there's some mileage here for simplification of the process (so long as installing mbed-ls is straightforward)

Neon22, 2015-12-03:
The relevant lines in dmesg output are all in one adjacent block:
```
[50797.130181] usb-storage 2-7:1.0: USB Mass Storage device detected
[50797.130469] scsi host5: usb-storage 2-7:1.0
[50797.149522] hid-generic 0003:0D28:0204.0005: hiddev0,hidraw1: USB HID v1.00 Device [MBED MBED CMSIS-DAP] on usb-0000:00:0b.0-7/input3
[50797.151518] cdc_acm 2-7:1.1: ttyACM0: USB ACM device
[50798.134129] scsi 5:0:0:0: Direct-Access     MBED     DAPLINK VFS      0.1  PQ: 0 ANSI: 2
[50798.134978] sd 5:0:0:0: Attached scsi generic sg2 type 0
[50798.147487] sd 5:0:0:0: [sdb] 16512 512-byte logical blocks: (8.45 MB/8.06 MiB)
[50798.152514] sd 5:0:0:0: [sdb] Write Protect is off
[50798.152532] sd 5:0:0:0: [sdb] Mode Sense: 03 00 00 00
[50798.156222] sd 5:0:0:0: [sdb] No Caching mode page found
[50798.156237] sd 5:0:0:0: [sdb] Assuming drive cache: write through
[50798.189175]  sdb:
```
TTY:
- perhaps  grepping for "USB HID" and "MBED CMSIS-DAP" on the same line followed by "tty*" on the next line would extract the ttyACM0 port identifier.

SD card:
- perhaps grepping for "MBED     DAPLINK VFS" and then looking on severalsubsequent lines for "[sd*]" would identify the mount point for the sd card

deshipu, 2015-12-03:
The problem is, if the user has connected other devices before and/or after, how do you know which one is yours?

Neon22, 2015-12-03:
It appears the (https://github.com/ARMmbed/mbed-ls) project has same issue.
I.e. if you connect multiple microbits - how do you make sure the tty you're talking to is the same as the one associated with the sdcard. There is a serial number in an earlier part of the dmesg.

I think they only way you can know is by adjacency in the output file. They do not appear to be processed in parallel (but I only have one Microbit).
In terms of "Is it a Microbit ?" - the MBED CMSIS-DAP identifies it as pretty much likely to be one.

The problems of an automatic tool instead of a procedure to follow....

remay, 2015-12-04:
I haven't re-checked, but I thought the mbed-ls implementation jumped through hoops to ensure that it reports the correct mapping between device, flash drive mount point and serial device.   Specifically I beleive that it can cope with multiple simultaneously connected devices - if I remember correctly from when I last looked at the source it goes grubbing deep in the OS device details (possible using some of the PnP ids?) to do this - and there are implementations for all the main OS's.  I believe that it deserves some further due dilligence ...

I beleive that it also digs an ID out of the html file found in the root of the mbed flash drive that will positively identify the device as a micro:bit (If you're running a recent firmware then that html file points to the mbed platform page for the device: https://developer.mbed.org/platforms/Microbit/ - the latest firmware and instructions for applying it are at the bottom of the page).

"MBED CMISS-DAP" is generic and applies to any mbed platform with the CMISS-DAP USB protocol - whilst for our users they are unlikely to have another device that identifies itself like this there may be cases where users want to attach more than one micro:bit to a single PC.

If I wasn't travelling I'd give this a bit more time right now to actually confirm my memories from when I last dug through the code.

dpgeorge, 2021-09-06:
As of c199d6bb22330cfeb76c48c8d295eb0b3dcfe8c5 the docs mention "sudo pip3 install yotta".

closed 2021-09-06 by dpgeorge