bbcmicrobit/micropython #115: What does Image.slice do?

ntoll, 2015-11-28:
I can guess, but it returns a type of `SlicedImage` which can't be displayed, has no methods and appears to be otherwise useless for kids. :-/  

I'm guessing this is something that may need to be hidden from kids given that it'd cause confusion? Advice most welcome!

(Context: I'm documenting the undocumented things for the `help` builtin function. I've had a lot of fun playing with the other methods.)

willingc, 2015-11-28:
@ntoll Feel free to use what you need from PR #65 for the ``help`` function. Not sure what the hold up is on merging the PR.

ntoll, 2015-11-28:
@willingc I've stacked it on top of #65. ;-)

Only got the music module to do then I'll create a PR.

markshannon, 2015-11-29:
`def slice(start, width, stride) returns an iterable of images`

`SlicedImage` is an iterable of Images. It has one important method: `__iter__` which returns an iterator of images. 

```python
>>> def sweep(img):
...     for x, i in enumerate(img.slice(0,1,1)):
...         yield(i.crop(-x, 0, 5, 5))
```
`display.animate(sweep(Image.PACMAN), 100)`

ntoll, 2015-11-29:
Aha! I see what it does now. I guess I was bitten by #112 when exploring the resulting iterator (`__iter__` is not listed if you `dir` the object).

deshipu, 2015-11-29:
That is a super-weird name for such a method, especially considering that Python has slices itself, also related with iteration, but doing something completely different.

markshannon, 2015-11-29:
When I implemented it i was thinking of `Image.slice` as just an internal implementation detail of `display.animate`. 
However, given that `display.animate` is being killed off https://github.com/bbcmicrobit/micropython/issues/104,`Image.slice` does need to be public. 
Suggestions for a better name?

deshipu, 2015-11-29:
I'm not entirely sure what it does. If it just gives you an iterator with the same image repeated, then maybe ``Image.repeat(n)``?

deshipu, 2015-11-29:
By the way, it just occurred to me that we have no itertools module, and in particular, no ``itertools.chain``.

ntoll, 2015-11-29:
OK... so I've been racking my brains as to how to explain this to an 11yo. I'm also a tad confused.

@markshannon, you say that objects created by the function have one method (`__iter__`), but when I try to call it "by hand" weird things happen:

```
>>> foo = Image.PACMAN.slice(0, 1, 1)
dir(foo)
[]
>> foo.__iter__()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'SlicedImage' object has no attribute '__iter__'
>>> enumerate(foo).__next__()
(0, Image(' \n9\n9\n9\n \n'))
```

Not what I was expecting..! I was expecting to be able to call `__iter__` and get an iterator to poke with a stick (a `next` looking stick). ;-)

So, what does slice do? More importantly, what is slice *for*..? (What problem does it solve?)

deshipu, 2015-11-29:
```
>>> slice = microbit.Image.HEART.slice(0, 1, 1)
>>> microbit.display.show(next(slice))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: 'SlicedImage' object is not an iterator
```

ntoll, 2015-11-29:
It's odd that:

```
>>> next(iter(foo))
Image(' \n9\n9\n9\n \n')
```

... works!

markshannon, 2015-11-29:
That `Image.PACMAN.slice(0, 1, 1).__iter__()` doesn't work is a MicroPython bug.
`iter(Image.PACMAN.slice(0, 1, 1))` works, so calling `__iter__` directly should work.

What does it do? It slices an image into slices :smile: 
Each image in the iterable is of formed of a window as high as the original and the given width. The first image has an `x` offset of `start`, the second at `start` + `stride` and so on, until the window no longer overlaps with the original.
The original purpose of `Image.slice` was to implement `display.animate`:
```python
def animate(img, delay, stride, start):
    display.show(img.slice(start, 5, stride), delay)
```

markshannon, 2015-11-29:
@deshipu `SlicedImage' is not an iterator, it is an *iterable*. 
You need to use `iter(Image.HEART.slice(0, 1, 1))` or just use a `for` loop.

willingc, 2015-11-29:
@markshannon @ntoll I could see using ``Image.slice()`` to take out a chunk of the string representation of the Image.

deshipu, 2015-11-29:
Is there also an equivalent method for getting horizontal strips?

markshannon, 2015-11-29:
No, but `Image.slice` isn't really for chopping up images. Use `Image.crop` for that.
`Image.slice` was designed purely to implement `display.animate`. Given the confusion it seems to be causing, I think that deleting it is the best thing.

ntoll, 2015-11-29:
@markshannon I certainly believe it'd be *hard* for your average 11yo to grok what it was doing. I'm certainly struggling to explain it in a couple of sentences using only common words of no longer than two syllables. ;-)

ntoll, 2015-11-29:
If we remove confusion by removing `slice` then I believe we should remove it. Especially when the new `show` method will do all the funky animation kids will be interested in (rather than having to learn about `slice`). Happy to be shouted down.

deshipu, 2015-11-29:
Do you think it would be possible to actually implement Python's slicing protocol to make something similar? In Python, I would do that my adding ``__getitem__`` that takes tuples of ``(x, y)`` as indices:

```
>>> image[2, 3] # return the value of the pixel at (2, 3)
8
>>> image[0, 1: 3, 4] # return a sub-image, with upper left corner at (0, 1) and lower right at (3, 4)
<Image>
>>> image[::2, 2] # return an image created by removing every second row and column
<Image>
>>> image[None, 0] # return the first row of the image
<Image>
>>> image[2, None: 4:None] # return the columns from 2 to 4
<Image>
>>> image[::None, -1] # return an up-side-down image
<Image>
```

What do you think?

markshannon, 2015-11-29:
It is consistent with `numpy`, but too complex for children, I would think.
`Image.crop`, `Image.rotate` and `Image.flip` should be sufficient.
I'm not sure what happened to `Image.rotate` and `Image.flip` they were on the todo list.

`Image.crop` by itself is quite powerful, for example:
```python
>>> for i in range(-4,6):                     
...     display.show(Image.HEART.crop(i,i,5,5))
...     sleep(200)
```
scrolls the image diagonally across the display from bottom right to top left.

dpgeorge, 2015-11-29:
Image.slice should not be a replacement for animations, it's too complex.  The kids need to be able to do fun things (like animations) very easily.  Doing fancy image manipulation (like a loop with crop) is for advanced use.

markshannon, 2015-11-29:
@dpgeorge I don't think anyone was proposing that `Image.slice` be a replacement for anything. 
It just happens to be the way that `display.animate` (the version that takes stride and start) is currently implemented.
I think we have all agreed to delete it.

dpgeorge, 2015-11-29:
> I think we have all agreed to delete it.

Ok!

ntoll, 2015-11-30:
+1 I think this is the way to go. Can't wait to see the updated API in action.

markshannon, 2016-01-21:
With #154, all trace of image slice has now gone.

closed 2016-01-21 by markshannon