bbcmicrobit/micropython #117: Higher level I²C interface

deshipu, 2015-11-29:
This is not a critical issue, I suppose, but I thought I will report it so that I don't forget about it.

Our I²C interface only has two commands, read bytes and write bytes. While this allows us to do everything that is needed (especially since we have control over the stop of transaction), it's a very low-level interface. If we want this to be actually used, it would be nice to have a higher level interface.

The most popular Python library for doing I²C communication is the ``smbus`` module. It's commonly used on the Raspberry Pi, and there are many tutorials online and examples of communicating with various devices using it. I think that if we mirrored the interface of that library (which in turn mirrors the interface of the C library that it wraps), we could count on reusing all of those examples, which is a big advantage.

I couldn't find any proper documentation of the library, but the docstrings in the code are explanatory:
https://github.com/bivab/smbus-cffi/blob/master/smbus/smbus.py

carlosperate, 2015-11-29:
Have the toch develop guys already implemented I2C/SPI/UART? While I do like the idea of having the same API as popular library, it's worth considering what the users might get familiarised with first while using the other high level microbit platforms.

markshannon, 2016-11-07:
smbus looks like a lot of code for our tiny device. 
Anyone volunteering to add a cut-down version to the `examples/` folder?

deshipu, 2016-11-07:
After some discussions about the `machine.I2C` interface in the upstream micropython, I think it would be good to implement an "i2c endpoint" class for this. I will take a stab at this.

deshipu, 2016-11-07:
Note that this will have an additional advantage -- any drivers created for regular micropython i2c devices will work without modification on the microbit.

deshipu, 2016-11-07:
Since the specification for the i2c endpoints is still a little unclear, I went and implement the SMBus commands after all. They should be familiar for many from the Raspberry Pi.

closed 2021-04-02 by deshipu