bbcmicrobit/micropython #119: microbit.i2c.write() does nothing

deshipu, 2015-11-29:
So, today I tried to connect the micro:bit to my quadruped robot (http://tote.rtfd.org) using I²C. I have a firmware (https://bitbucket.org/thesheep/tote/src/tip/i2cslave/) that makes the Arduino Pro Mini on that robot behave like a simple servo controller -- it basically just listens for the servo positions on the I²C bus.

I have previously tested this using Raspberry Pi and the following code:
```
import smbus
import struct
import math

bus = smbus.SMBus(1)
ADDRESS = 0x04

def servo_move(servo, radians=None, degrees=None):
    if radians is None:
        radians = math.radians(degrees)
    bus.write_block_data(ADDRESS, servo,
        [ord(b) for b in struct.pack('f', radians)])

servo_move(0, degrees=45)
```
so I know it is working properly. Now, I tried this with micro:bit:
```
import ustruct
import math
from microbit import i2c

i2c.write(0x04, ustruct('bf', 0, math.radians(45)))
```
The effect? Nothing at all. So I thought that maybe I don't understand how ``write_block_data`` works. I decided to make a simpler test. I wrote the following Arduino sketch:
```
#include <Wire.h>

void setup() {
  Serial.begin(115200);
  Wire.begin(8);
  Wire.onReceive(receiveEvent);
  Serial.println("Listening...");
}

void loop() {
  delay(100);
}

void receiveEvent(int howMany) {
  while (0 < Wire.available()) {
    char c = Wire.read();
    Serial.print(c, HEX);
  }
  Serial.println();
}
```
And I tested it with the following Python code on the Raspberry Pi:
```
import smbus

bus = smbus.SMBus(1)
bus.write_byte(8, 10)
```
It correctly printed "A". Now, trying that with micro:bit:
```
from microbit import i2c

i2c.write(8, b"X")
```
does nothing. Note, that it does nothing no matter if the Arduino is connected or not -- it doesn't even raise any exceptions if the I²C bus is not connected at all.

I suppose that concludes my work on getting this robot to work with micro:bit.

markshannon, 2015-11-29:
There are two issues here.
1. We should raise an exception if the underlying DAL method fails.
2. If the underlying write method doesn't work, then it needs to be reported as a DAL issue to 
https://github.com/lancaster-university/microbit-dal/issues

markshannon, 2015-11-29:
Could you set the pins that communicate with the robot directly?

deshipu, 2015-11-29:
> Could you set the pins that communicate with the robot directly?

I don't understand what you mean here.

deshipu, 2015-11-29:
I just now tried this with additional 10kΩ pullup resistors. Same effect.

deshipu, 2015-11-29:
Reported it at https://github.com/lancaster-university/microbit-dal/issues/54

carlosperate, 2015-11-29:
>  > Could you set the pins that communicate with the robot directly?

>I don't understand what you mean here.

I believe he means if you can set the those pin states (digital or analogue), just to ensure that there is not an issue with the pin themselves.

carlosperate, 2015-11-29:
> I just now tried this with additional 10kΩ pullup resistors. Same effect.
Are you able to check the I2C lines with an oscilloscope? Also, what is the 7bit address that the device responds to? Remember that the LSB will be set to read or write and sometimes the documentation will give you an full 8bit address for read, and another for write.

markshannon, 2015-11-29:
I note that you are using 115.2k baud for the clock speed on your slave device. It looks like the mbed I2C runs at 100k baud:
microbit-micropython/yotta_modules/mbed-classic/common/I2C.cpp

carlosperate, 2015-11-29:
The `Serial.begin(115200);` would be for the uart comms, not the I2C. I believe the arduino wire library operates at 100kHz by default.

dpgeorge, 2015-11-29:
Maybe try communicating with the onboard sensors (accel, compass) over I2C because we know they definitely work.

deshipu, 2015-11-29:
@carlosperate I don't have access to an osciloscope right now, unfortunately. The address is 4 in case of the robot, and 8 in case of the simple byte reading test.
@markshannon I'm not an specialist, but I thought that the speed is up to the master, as it is handling the clock line, and sometimes the slaves can slow it down by holding the clock down. I certainly see no way of setting the baud rate for I2C in the Arduino Wire library. That ``Serial.begin()`` is for the serial baud rate, which I use for communicating back with the computer.
@dpgeorge I already packed it all up for the next person on the Micro World Tour, but I can unpack and try. What are the addresses of those devices? Are they on the same I2C bus?

If they are on the same bus, we should probably have a warning in the documentation saying that using those pins for other purposes will make the accelerometer and compass unavailable?

deshipu, 2015-11-29:
I tried the newest build with the error handling, and I can see I always get the same error, no matter what is connected to the I2C pins:
```
>>> i2c.write(0x04, ustruct.pack('bf', 0, math.radians(45)))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: I2C write failed with error code -1010
```
Not sure where to look for the error codes table...

markshannon, 2015-11-29:
microbit-micropython/yotta_modules/microbit-dal/inc/ErrorNo.h

dpgeorge, 2015-11-29:
The accel is address 0x3a, the compass, 0x1d.

The accel and compass use the same DAL I2C functions that the Python I2C binding expose, and they work, so I don't know what the problem is.

deshipu, 2015-11-29:
``MICROBIT_I2C_ERROR = -1010``

Sigh :(

deshipu, 2015-11-29:
@dpgeorge thank you. I can confirm that writes both to 0x3a and 0x1d work.

The question that comes into my mind -- is this the same i2c bus as the one exposed on pins 19 and 20?

dpgeorge, 2015-11-29:
> The question that comes into my mind -- is this the same i2c bus as the one exposed on pins 19 and 20?

Yes: SDA = P20 and SCL = P19.

carlosperate, 2015-11-29:
Maybe try different value resistors, the capacitance in the line will be dependant on your set up, so you might need to play a bit with that. It would be a lot easier with an oscilloscope as you could see the effects directly, but a quick trick could be to use a variable resistor/potentiometer and have a loop trying to talk to your arduino, so you can easily the successful transmission once you reach an appropriate value. You also have to keep in mind that if there is already a pull up resistor in the microbit (I don't think the schematics are available, and we cannot check the DAL source code to see if the internal pull might be configured), adding another one in parallel would basically create a current divider, so increasing the value of your external resistor would reduce the total pull resistance.

deshipu, 2015-11-30:
I got it to work. It turns out, that the address the DAL functions expect is 8bit -- that is, you have to shift your normal address one bit to the right: ```i2c.write(ADDRESS << 1, bytes)```.

I wonder if we should do that for the user in our wrappers, or just document it clearly?

carlosperate, 2015-11-30:
> I wonder if we should do that for the user in our wrappers, or just document it clearly?

If the DAL doesn't do it, we definitely should. If we are providing a "write" and "read" methods, the most logical assumption is that the input argument expects the 7bit address and automatically adds the LSB accordingly.
However, because the DAL offers the same interface (MicroBitI2C.read and MicroBitI2C.write), this is perhaps something that should be implemented there?

ntoll, 2015-11-30:
@carlosperate it's already been suggested upstream. Let's see what @finneyj and @jamesadevine make of it (the DAL is their baby). :-)

dpgeorge, 2015-11-30:
I agree that we should do the left shift behind the scenes.  The address passed in should be the 7-bit address of the device.  FWIW, upstream MicroPython does the left shifting for you.

dpgeorge, 2016-02-01:
Should we left shift the address?

carlosperate, 2016-02-01:
> Should we left shift the address?

Do you mean it is not going to be implemented on the DAL side?
I still believe it should be transparent for the user, if the user is offered a read or write function, they would expect to use the exactly same address for both of them, without having to modify it.

deshipu, 2016-02-01:
Those functions modify the address for reading/writing internally -- they set the least significant bit to 0 or 1, overwriting whatever you actually passed. So it's just a question of shifting it.

dpgeorge, 2016-02-01:
No, the DAL doesn't do the shifting, we need to.

carlosperate, 2016-02-01:
Then I would still vote for shifting on the micropython side and specified a 7 bit address. I would expect to be able to use the same address value for both write and read without further modifications on the user side.

carlosperate, 2016-03-07:
Just a quick note for the mods, it looks like this issue can be closed manually, as the https://github.com/bbcmicrobit/micropython/issues/186 merge has not done so automatically.

closed 2016-03-07 by deshipu