bbcmicrobit/micropython #124: Create setup/on-boarding documentation

ntoll, 2015-11-30:
Must take into account the following platforms:

  - [ ] Windows
  - [ ] RaspberryPi
  - [ ] Mac OSX
  - [ ] Ubuntu / Debian
  - [ ] Vanilla Linux

Can include screen-shots, links and videos if required.

**MUST HAVE**

  - [ ] Plugging in the device and flashing Python code onto it
  - [ ] Connecting to the REPL

**Nice to have**

  - [ ] Setting up a development environment for MicroPython development
  - [ ] Using command line tools to generate .hex files to flash onto the device.

willingc, 2015-12-01:
@ntoll I put a skeleton doc structure for this together. If we like it and merge it, others can help flesh out the content. Will open a PR.

Here's a screenshot of the sidebar navigation and one doc.
[image]

ntoll, 2015-12-01:
@willingc great minds think alike, or fools seldom differ. I was just about to start doing something about this, this evening only to find you're several steps ahead of me! Also, I've just replied to an email from Naomi where she's having similar thoughts.

Three heads are better than one! :-)

ntoll, 2015-12-01:
Also, I've been working on some sort of Windows solution for teachers based on a Chromium App. The workshops have demonstrated it's quite a painful journey and I think this could be one way to help given the prevalence of both Chrome (the browser) and Chromebooks in classrooms. :-/

willingc, 2015-12-01:
No problem. I find good docs help with making things more pleasant to use.

I just got a Windows 10 desktop so I will be able play around with it. Please encourage folks to iterate on the docs :D

ntoll, 2015-12-01:
@willingc you have an account on RtD? I can add you as an admin!

willingc, 2015-12-01:
@ntoll Yep, I do (same as my github username). Not sure why the docs didn't autodeploy without hitting build in RTD. Probably don't have the hook set up in the RTD settings.

willingc, 2015-12-01:
GIFs are a good way to give people an overview of steps in the process. Most folks can only retain 3 or 4 steps at a time.

ntoll, 2015-12-01:
@willingc OK... you're added as an admin.

willingc, 2015-12-01:
@ntoll Thanks.