bbcmicrobit/micropython #126: Weird behavior of struct.pack

deshipu, 2015-11-30:
So once I got I2C working, I started to make use of struct.pack to send the right bytes, and I stumbled upon this:

```
>>> import struct
>>> struct.pack("f", 0.1)
b'\xcc\xcc\xcc='
>>> struct.pack("<f", 0.1)
b'\xcc\xcc\xcc='
>>> struct.pack("<ccf", 0, 4, 0.1)
b'\xcc\xcc\xcc='
>>> struct.pack("<c", 0)          
b''
```

dpgeorge, 2015-11-30:
The 'c' code is not implemented.  Use 'B' instead (and then the arg should be an integer, so use ord(bytes) if needed).

deshipu, 2015-11-30:
ok, one more weirdness:
```
>>> sp(f0.t.pack("b<f", 0, 0.1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can't convert float to int
>>> struct.pack("b<f", 0, 0xff, 0.1)
b'\x00\x00\x00\x00\xcc\xcc\xcc='
```

deshipu, 2015-11-30:
(I know that "b<f" is invalid, but still)

dpgeorge, 2015-11-30:
If you give it a valid format string (excl using 'c') then it'll work.  And it's not crashing when the format string is invalid, so that's good enough.

If you want to make all these corner cases given decent error messages then MicroPython grows 2x in size and no longer works on microbit :(

deshipu, 2015-11-30:
Understood.

closed 2015-11-30 by deshipu