bbcmicrobit/micropython #127: Building old version?

jrmhaig, 2015-12-01:
I have recently rebuilt and when I upload it appears to result in an old version. Specifically, the help documentation does not have the 'Stuff to explore' section and I have found that the 'music' functions are missing.

I have checked out revision 0189136af557bc2c8514935f7351bcaafe32c47f (Allow music tunes to be played in the background) as this was the actual update I am interested in and this is working correctly for me.

ntoll, 2015-12-01:
@jrmhaig all of these changes should be in master. Just pulling latest master should fix it. :-)

dpgeorge, 2015-12-02:
Music is now a separate module and you need to do `import music`.

jrmhaig, 2015-12-03:
OK, it was music going into a separate module that confused me. Sorry.

closed 2015-12-03 by jrmhaig