bbcmicrobit/micropython #132: Consider exposing USB_VCP module.

Lukasa, 2015-12-10:
Given that we have a substantial bit of code based around communicating using the serial-over-USB interface into the Micro:bit, it's a little bit perplexing that Python code cannot actually use that interface, and instead must connect wires to pins. Is there any reason we cannot expose the serial-over-USB interface directly from MicroPython?

ntoll, 2015-12-10:
To quote Damien from an earlier thread... is this perhaps useful..?

Serial port:

There is a uart object to read/write bytes to the serial port, using `uart.read(nbytes)` and `uart.write('string')`. They will interfere with the REPL but maybe that's what you want. Eg:

```
while True:
    if uart.any():
        print(uart.read(1))
````

You can assign the uart to use pin0, pin1 as tx/rx (or any other pins)
but then you lose the REPL. Reassign using:

```
uart.init(pins=(pin0, pin1)
```

Lukasa, 2015-12-10:
So, to be clear, when we say `pin0` and `pin1`, I assume those do not refer to serial-over-USB, but instead refer to the literal micro:bit pins. This totally works and I'm fine with that, but it's a little perplexing that we can't do this over the USB interface given that the REPL itself can.

ntoll, 2015-12-10:
Right... you're correct on both counts (pins and it's perplexing about serial over USB). That's one for @dpgeorge although the above will give you *something* to start with when it comes to serial communication, if you see what I mean. Hopefully, @dpgeorge can sort you out with the USB side of things.

dpgeorge, 2015-12-10:
> Given that we have a substantial bit of code based around communicating using the serial-over-USB interface into the Micro:bit, it's a little bit perplexing that Python code cannot actually use that [...]

As @ntoll says, the USB serial is available using the uart object.  uart.read and uart.write.

> So, to be clear, when we say pin0 and pin1, I assume those do not refer to serial-over-USB, but instead refer to the literal micro:bit pins.

Yes.

> This totally works and I'm fine with that, but it's a little perplexing that we can't do this over the USB interface given that the REPL itself can.

I'm not sure what you mean here...?

Let me explain a few things.  There are internal pins connected from the nRF51 chip to the USB-uart converter.  These pins are not available from the Python side (simply because we didn't add them, they could easily be added as pin_tx and pin_rx, for example).  The uart (there is only 1 uart) can be configured to output/input on any set of pins, and on boot up it's configured to output/input on the internal pin_tx/pin_rx pair.  So if you do uart.read/uart.write you communicate with the USB-uart converter and hence the PC.  If you want the uart to go to pin0/pin1 then you can use uart.init to do that, but then you lose the REPL.

@Lukasa what exactly do you need/want?

One thing we can do is make sys.stdin and sys.stdout available.  They would provide another way to read/write from/to the USB serial.

dpgeorge, 2015-12-10:
Just to clarify further.  Regarding the title of this issue "USB_VCP module": there is no USB peripheral on the nRF51 chip.  That all happens on the support chip, which we don't have access to, so we can't expose anything to do with USB_VCP (all comms is done through the nRF51's uart).

Lukasa, 2015-12-11:
@dpgeorge So, to be clear, the configuration of uart to do I/O on the pin_tx, pin_rx pair is generally true when running MicroPython, and is not specifically true when running the repl firmware?

If that's the case, I need to do a bit more work to understand why my serial communication is falling over. Seems like the documentation misled me here (it doesn't talk about the fact that the uart module is already initiated to use serial-over-USB).

deshipu, 2015-12-11:
Perhaps instead of exposing the pin_tx and pin_rx, we could just make them the defaults for uart.init()?
This way we are not littering the namespace, but still can get back to the initial setup.

ntoll, 2015-12-11:
@deshipu less is more +1 ;-)

markshannon, 2016-11-07:
Do we consider this to be fixed by https://github.com/bbcmicrobit/micropython/pull/244?
That appears to implemented @deshipu's suggestion.
@Lukasa anything else?

Lukasa, 2016-11-07:
Nothing else from me!

closed 2016-11-08 by dpgeorge