bbcmicrobit/micropython #137: write_analog doesn't accept a float

tomviner, 2015-12-13:
The [docs say](https://microbit-micropython.readthedocs.org/en/latest/pin.html#microbit.MicroBitAnalogDigitalPin.write_analog):
> write_analog(value)
> Output a PWM signal on the pin, with the duty cycle proportional to the provided value. The value may be either an integer or a floating point number between 0 (0% duty cycle) and 1023 (100% duty).

But:
```python
>>> microbit.pin0.write_analog(0.5)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can't convert float to int
```

Let's decide if it should be allowed or not, and fix either the docs (which I can do a patch for) or the implementation.

carlosperate, 2015-12-13:
I also believe it should accept floating point, as a lot of people will input the result of a mathematical operation. Of course, the peripheral requires an integer, but that is a conversion that should be done inside the micropython function,

dpgeorge, 2015-12-13:
Analog output is in principle a "continuous" quantity, so I'd say it should accept a float.

tomviner, 2015-12-13:
Docs fix: https://github.com/bbcmicrobit/micropython/pull/138

I'll leave getting floats working for someone with better c++ knowledge.

PR #150 (2016-01-13): Various fixes
There are 4 small commits here to fix various problems.  See message of each one for more detail.

Fixes #137 and #147.

@ntoll if you don't mind merging :)
closed 2016-01-13 by markshannon