bbcmicrobit/micropython #140: Music on multiple pins

jrmhaig, 2015-12-19:
This is probably going way off the beaten track here and maybe the answer is "it probably cannot be done".

I would like to play music on multiple pins but if I try:

```python
music.play('a:30', microbit.pin0)
music.play('b:30', microbit.pin1)
```

it plays the second note on both pins. If I use `music.pitch` instead of `music.play` (which is what I really want because I'm being even more contrary) the note turns into a distorted screech.

ntoll, 2015-12-19:
The simple answer is "it probably cannot be done". I seem to recollect that polyphony (two or more notes at once) was either not possible, or so fiddly to get done that it wasn't (yet) worth the effort. If the latter, I guess it'd be prioritised depending on user feedback and feature requests.

Ask me sometime about my micro:bit orchestra. ;-)

jrmhaig, 2015-12-19:
*Ahem* Consider this a feature request (and I'm hoping it is possible but too fiddly). :-)

I happen to have found a score for Ron Grainer's Dr Who theme and I think that Delia Derbyshire would have approved. For the moment I'll have to see how well I can sync multiple bits.

dpgeorge, 2015-12-21:
Anything is possible :)  In this case we'd need to steal a second hardware timer to get an independent PWM channel.  From memory that is actually possible.

DavidWhaleMEF, 2017-05-12:
Note that through BBC learning, we recently surfaced an approved arrangement of Ron Grainer's Doctor Who Theme via the Doctor Who activities - in both PXT and Python. No polyphonics in this case but it sounds pretty good. I was thinking we could use @ntoll trick of radio sync between two micro:bits to get the accompaniment running on a second micro:bit. It's in the 'secret area' once you hack into the Dalek mothership.

jrmhaig, 2017-05-12:
Erm, you've seen this? :-)

https://www.youtube.com/watch?v=uLIaXOPBD_M

Officially "Awesomeness on stilts" (according to @ntoll)

DavidWhaleMEF, 2017-05-12:
Oh yes, I remember seeing that at the time you did it, but I must admit we all forgot about it!

Have you surfaced the code for this, and if so, would you mind sharing the link to it?

ntoll, 2017-05-12:
For reference: https://gist.github.com/ntoll/f762c1e2c06104854b6142df4cb3c095