bbcmicrobit/micropython #143: "Missing PLATFORM_SPECIFIC_HAL_LIB" when attempting to compile

ncoghlan, 2015-12-27:
I'm aiming to create a Docker image so folks with Docker appropriately configured can just do `docker run -it --device=/dev/ttyACM0 --name=microbit ncoghlan/microbit-dev` to get a mostly configured build environment regardless of their preferred desktop OS.

The Dockerfile and README are at https://github.com/ncoghlan/microbit/tree/master/devenv

However, running the "yt build" step in a checkout of the bbcmicrobit/micropython repo in the resulting container currently leads to the following:

    $ yt build
    info: generate for target: bbc-microbit-classic-gcc-nosd 0.1.2 at /home/microbitdev/micropython/yotta_targets/bbc-microbit-classic-gcc-nosd
    GCC version is: 4.9.3
    suppressing warnings from nrf51-sdk
    CMake Error at /home/microbitdev/micropython/yotta_modules/mbed-hal-nrf51822-mcu/CMakeLists.txt:94 (message):
      Missing PLATFORM_SPECIFIC_HAL_LIB

    -- Configuring incomplete, errors occurred!
    See also "/home/microbitdev/micropython/build/bbc-microbit-classic-gcc-nosd/CMakeFiles/CMakeOutput.log".
    error: command ['cmake', '-D', 'CMAKE_BUILD_TYPE=RelWithDebInfo', '-G', 'Ninja', '.'] failed

I get the same error with `bbc-microbit-classic-gcc` as the build target.

Searching for PLATFORM_SPECIFIC_HAL_LIB brings up http://forums.mbed.com/t/nrf51822-mkit-build-fails-with-missing-platform-specific-hal-lib/807/6, which suggests that this can be caused by the "mbed-target" keyword not being set in target.json, but that doesn't appear to be the case for either of these targets.

dpgeorge, 2015-12-27:
I can confirm the issue.  It just did a `yt update` and a lot of new yotta packages/modules were downloaded (and even required me to log in to mbed...) and now `yt build` fails for me with `Missing PLATFORM_SPECIFIC_HAL_LIB` as above.  So looks like a fresh bug coming from an upstream configuration change.

ncoghlan, 2016-01-01:
I'm not sure if it's related (since I've never had a working configuration), but I get a complaint from yotta's attempting to figure out the derived target if I query the current target:

    $ yt target
    Fatal Exception, yotta=0.12.2
    Traceback (most recent call last):
      File "/home/microbitdev/.local/bin/yt", line 4, in <module>
        yotta.main()
      File "/home/microbitdev/.local/lib/python3.4/site-packages/yotta/main.py", line 205, in main
        status = args.command(args, following_args)
      File "/home/microbitdev/.local/lib/python3.4/site-packages/yotta/target.py", line 100, in execCommand
        return displayCurrentTarget(args)
      File "/home/microbitdev/.local/lib/python3.4/site-packages/yotta/target.py", line 66, in displayCurrentTarget
        args.target, c.targetsPath(), application_dir=app_path, install_missing=False
      File "/home/microbitdev/.local/lib/python3.4/site-packages/yotta/lib/target.py", line 155, in getDerivedTarget
        assert(isinstance(t, Target))
    AssertionError

ncoghlan, 2016-01-01:
I tracked down the origin of the `yt build` problem and filed a bug against yotta: https://github.com/ARMmbed/yotta/issues/643

ncoghlan, 2016-01-01:
(Commenting out the assert that was causing `yt target` didn't help at all with `yt build`)

ncoghlan, 2016-01-01:
To attempt to make progress, I commented out the failing assertion (which was added back in September): https://github.com/ARMmbed/mbed-hal-nrf51822-mcu/commit/2cbe8354659041a327346b8a5fb7edc3450dff15

That allowed further progress, but one of the build steps wanted Python 2, which isn't currently part of the Docker image. I'll add that and try again.

ncoghlan, 2016-01-01:
Ignoring the warning got me similar symptoms to https://github.com/ARMmbed/mbed-drivers/issues/108 which suggests there is indeed something wrong with the current target definitions.

ncoghlan, 2016-01-01:
The Lancaster University folks were able to help get the issue resolved via https://github.com/lancaster-university/microbit-targets/issues/1, so I while I don't have a working build yet (see the last message on that issue for details), at least this particular problem has been resolved.

closed 2016-01-01 by ncoghlan