bbcmicrobit/micropython #147: BUG - random(-n) always returns -1001

ntoll, 2016-01-09:
This is what I see:

```
random(-3)
-1001
```

I was expecting a number between 0 and -3.

whaleygeek, 2016-01-09:
I *think* -1001 is a DAL error code (all the error codes were re-mapped to
-1000 and below in a change before Christmas).

___________________________________________________________
David Whale, B.Sc (Hons), MIET
*Software Engineer and IET Schools Liaison Officer, Essex*

email:  dwhale@theiet.org
twitter: @whaleygeek
blog:  blog.whaleygeek.co.uk

Co-author of global bestselling book "Adventures in Minecraft"
<http://amzn.to/ZGfxZG> - lets get kids coding!
Now in English, Spanish, Chinese and Russian.

ntoll, 2016-01-09:
That's what I wondered. Personally, I think we should only accept positive integers and you negate the result if you want a minus number. Anything < 0 results in Python's `ValueError`.

dpgeorge, 2016-01-13:
Yes, it's due to the DAL rejecting inputs 0 or below.

I agree we should raise a ValueError for inputs 0 or below.

markshannon, 2016-01-13:
Fixed.

closed 2016-01-13 by markshannon