bbcmicrobit/micropython #148: pin0.is_touched() always returns False

ntoll, 2016-01-12:
If I'm touching the pin, it should return True.

ntoll, 2016-01-12:
This also applies to the other pins too.

ntoll, 2016-01-12:
Looking at https://github.com/bbcmicrobit/micropython/blob/master/source/microbit/microbitpin.cpp#L98 it's just a wrapper around a function in the DAL. :-/

dpgeorge, 2016-01-12:
I think you need to touch ground of the microbit with your other hand.  I'm pretty sure I got is_touched() working before.

whaleygeek, 2016-01-13:
Yes, we were using this at the club the other week, it does work, but you
do have to hold GND.

It's resistive touch, (not capacitive touch as the original spec blog page
says), so it needs a path to ground in order to work.

This just about works between two people if the grounds are close and on
the same arms as touching (see the Love-ometer resource that the BBC
created using this).

https://www.microbit.co.uk/td/lessons/love-meter

The front picture on the video in the above link shows how you have to make
the circuit for it to work.

___________________________________________________________
David Whale, B.Sc (Hons), MIET
*Software Engineer and IET Schools Liaison Officer, Essex*

email:  dwhale@theiet.org
twitter: @whaleygeek
blog:  blog.whaleygeek.co.uk

Co-author of global bestselling book "Adventures in Minecraft"
<http://amzn.to/ZGfxZG> - lets get kids coding!
Now in English, Spanish, Chinese and Russian.

ntoll, 2016-01-13:
@dpgeorge @whaleygeek I've been touching GND while holding pin0. Can't get anything other than False out of it.

whaleygeek, 2016-01-13:
What happens if you follow the touch develop resource I linked you to -
does that work? That worked for me a couple of weeks before Christmas,
although there has been another drop in the last few days that is
percolating it's way up.

i.e. is it a runtime problem, or a python problem?

Also, can you confirm what version of the runtime microPython is built
with, and what version of the runtime the latest TouchDevelop is built
with, in case of a difference there.

___________________________________________________________
David Whale, B.Sc (Hons), MIET
*Software Engineer and IET Schools Liaison Officer, Essex*

email:  dwhale@theiet.org
twitter: @whaleygeek
blog:  blog.whaleygeek.co.uk

Co-author of global bestselling book "Adventures in Minecraft"
<http://amzn.to/ZGfxZG> - lets get kids coding!
Now in English, Spanish, Chinese and Russian.

ntoll, 2016-01-13:
I'll check this evening... (currently at "real work" tm)

dpgeorge, 2016-01-13:
I can confirm that is_touched does not work on pin0, pin1 or pin2.  I tried with my hands, and with a 1k resistor (connecting pin to gnd) and neither worked.

MarcScott, 2016-01-13:
Also can confirm that this doesn't work with or without touching gnd

dpgeorge, 2016-01-13:
The problem is at our end.  The reason is that we (rightly or wrongly) hijack the system ticker and prevent the DAL from updating its internal components.  As such the touch detection does not work.

@markshannon this is going to be a recuring problem, where we are fighting against the DAL.  We really should be just providing a wrapper around the DAL and letting it do all the dirty work with events etc.

markshannon, 2016-01-13:
You are right, it is going to be a recurring problem. 
I don't want to fight against the DAL, but its authors are effectively forcing us to do so.
The refuse to deal with issues, accept PRs, or open source the code so we can fork it.
:disappointed:

BTW, all the DAL does for pins is provide trivial de-bouncing, the same as for the buttons.
I submitted a PR to the DAL project, a couple of months ago, to factor out the debugging logic. Still waiting...

ntoll, 2016-01-13:
HNY @markshannon! The project will be opened up, but it's the BBC who are blocking on this. Joe et al really want this to be open but can't make it so. (Welcome to my world - coping with BBC decisions that only make sense to them from an organizational rather than technical perspective). :-(

Any suggestions for next steps?

ntoll, 2016-01-19:
Please see my latest comment pm #152.

ntoll, 2016-01-19:
Please see my latest comment on #152 :-) (tl;dr - @ntoll is an idiot)

closed 2016-01-19 by ntoll