bbcmicrobit/micropython #153: Consistent handling of asynchonicity, repetition and iterables

markshannon, 2016-01-16:
The methods `display.animate`, `display.scroll` and `music.play` all consume iterables, to produce a sustained output (either visual or audio). They also have the option to perform asynchronously and to repeat the input.

Our current approach is inconsistent and partial.
`music.play` does not support asynchronous play.
`music.play` does not support arbitrary iterables, just lists and tuples.
`display.animate` crashes if passes a generator with `loop=True` :disappointed:
`display.animate` and `display.scroll` are not interruptable, which means that if passed `loop=True, wait=False`, then a hard reset is required.

What I propose:
* Add a repeat iterator (described below). 
** `f(iter, ..., loop=True)` should be equivalent to `f(repeat(iter), ...)`, where `f` is one of the methods listed above.
* Add an async API  to handle asynchronous and synchronous iteration in a consistent fashion, with robust handling of exceptions whether synchronous or not.
** This will give us one place to handle the issues listed above and to do it consistently and correctly.

## The **repeat** iterator
Add a `repeat` iterable which takes any iterable and repeats it. The behaviour of this iterable should differ from `itertools.cycle` in that it shouldn't record the values (we don't have the RAM to waste), but repeatedly iterate over the underlying iterable.
To do this it must take care to handle exhausted iterators, by stopping iteration if the underlying iterator is exhausted. Here is the proposed semantics:
```
def repeat(iterable):
    while True:
        iterator = iter(iterable)
        try:
            item = next(iterator)
        except StopIteration:
            return
        yield item
        yield from iterator
```

### Implementation and API:
The `repeat` iterator will be implemented in C. It will have a simple C API of a single function.
Initially it won't be exposed at the Python level.

## The async API
This will be a data structure and a few C function to handle (a)synchronous iteration.
This will be a C level API not be exposed at the Python level.

### Proposed C API
```
// Opaque data structure
typedef struct _async_t async_t;
// Callback function pointer type for each item 
typedef void (*per_item_func_ptr)(void *state, mp_obj_t item);
// Callback function pointer type for function call when iteration is complete
typedef void (*done_func_ptr)(void *state);
// Start iteration -- 'state' is whatever state the passed functions need to operate. 
void async_start(async_t *async, void *state, per_item_func_ptr per_item, finish_func_ptr finish, bool wait);
// Stop iteration
void async_stop(async_t *async, bool raise_pending);
// Test whether iteration is still running
void async_is_running(async_t *async);
```
### Implementation schedule
To do...

deshipu, 2016-01-16:
https://docs.python.org/2/library/itertools.html#itertools.repeat

markshannon, 2016-01-16:
Do you want add some detail to that comment? :wink: 
It is not `itertools.repeat`, but [itertools.cycle](https://docs.python.org/3/library/itertools.html#itertools.cycle) that the proposed `repeat` is closest to. It differs for the reason given above.

deshipu, 2016-01-16:
Perhaps we should keep things similar to actual Python and not reinvent our own standard library? This way it will be easier for those kids to later learn programming in "grown up" python.

markshannon, 2016-01-16:
The problem with that is that it needlessly duplicates the iterable in the common case that it is a list or tuple. We just don't have the RAM to spare for such things.

These are primarily internal objects for the `display` and `music` module, so there is nothing exposed at the Python level.

deshipu, 2016-01-16:
If the standard library ``itertools.cycle`` doesn't have special handling for lists and tuples, I'm sure it could be added, so that it doesn't copy them? If you insist on your own function here, I would strongly suggest that you pick a different name, to avoid confusion.

As to your Python code, it won't work the way you think. And exhausted iterator raises ``StopIteration`` no matter what it iterates over. And no, wrapping it in another iterator won't help. Have you tested it?

dpgeorge, 2016-01-16:
Note that music.play does already implement async behaviour.  One would need to see if this existing code can be worked into the proposed general async adaptor.

markshannon, 2016-01-16:
@deshipu First of all it is a C level construct, so it is invisible to any user of uPy. I don't see how that can cause confusion.
Yes, I have tested it. How do you think that I think that it should work?

markshannon, 2016-01-16:
@dpgeorge I hadn't realized that `music.play` now supports async behaviour. That's good.

I don't want to change the API with this, just make things more robust and consistent.

dpgeorge, 2016-01-16:
> I hadn't realized that music.play now supports async behaviour. That's good.
> 
> I don't want to change the API with this, just make things more robust and consistent.

I think that music and display have their own details for async mode that might not be easy to generalise.  And since there are only these 2 components that need async mode I don't think it's worth generalising.  We could make some helper functions to reduce code duplication (eg the ctrl-C handling for wait=True, see music module for how it's done) but I don't think we need a full blown async API pattern.

deshipu, 2016-01-16:
> Yes, I have tested it. How do you think that I think that it should work?

Because I have tested it too, and I can't believe what it does is actually what you want. Here is what it does:

1. Takes the first element from the passed iterable.
2. If the iterable is empty, raises ``StopIteration`` (that ``try``-``except`` basically does nothing, because ``return`` inside a generator function raises ``StopIteration`` anyways.
3. yields that item
4. yields the rest of the iterator, exhausting it.
5. Tries to repeat, at which point the iterator is exhausted, so it triggers 2. immediately.

markshannon, 2016-01-16:
I think you may have missed the line
```python
iterator = iter(iterable)
```
There is a reason for the `try` `except`block:
https://www.python.org/dev/peps/pep-0479/

markshannon, 2016-01-16:
@dpgeorge What is the difference between "some helper functions" and a "full blown async API pattern"?
You say tomato, I say tomato (that does work so well in print).
I think we both want the smallest amount of code that will work correctly. 
There are several corner cases to consider and I think it best to have the code in one place.

dpgeorge, 2016-01-16:
Async works for both display and music.  So I'd say just leave it for now because we have other things to do.

An example of a helper function could be one to wait in a loop checking for ctrl-C.  But now that I look into it I see that such a function would need the "full blown async API" that you propose.  So easiest to just adapt the music.play one to a specific one for display.

Well, if you have the time to code it all up then by all means go ahead!  It should be pretty easy to verify your design by seeing if it can be used to rewrite the music async code.

deshipu, 2016-01-16:
> I think you may have missed the line
>
> ```
> iterator = iter(iterable)
> ```

I didn't miss it. Wrapping an iterator in another iterator doesn't make the original iterator restart. Infact, it pretty much does nothing.

> There is a reason for the try exceptblock:
> https://www.python.org/dev/peps/pep-0479/

Hmm, are you saying that Micropython skipped the whole ``__future__`` thing and implemented the incompatible change directly?

markshannon, 2016-01-16:
On 16/01/16 22:49, Radomir Dopieralski wrote:
>     I think you may have missed the line
>
>     |iterator = iter(iterable) |
>
> I didn't miss it. Wrapping an iterator in another iterator doesn't make
> the original iterator restart. Infact, it pretty much does nothing.

Do please read up on the difference between an iterator and an iterable. 
It is rather key to the workings of the proposed `repeat` iterator.

> Hmm, are you saying that Micropython skipped the whole |__future__|
> thing and implemented the incompatible change directly?
No, but why have example code that will break in a couple of years?

deshipu, 2016-01-16:
> Do please read up on the difference between an iterator and an iterable.
> It is rather key to the workings of the proposed `repeat` iterator.

I know the difference. In particular, I know that an iterator is an iterable. Your code breaks as soon as you pass an iterator (or any other iterable that can be exhausted) to it.

markshannon, 2016-01-16:
What do you mean by "breaks"?

deshipu, 2016-01-16:
Iterates over it once and stops.

markshannon, 2016-01-16:
That is what it supposed to do. What else could it do?

deshipu, 2016-01-16:
I don't know what you would expect from ``loop=True`` or a function called ``repeat()``, but my non-native-speaker understanding of English suggests that it should repeat the sequence.

markshannon, 2016-01-16:
How do you repeat a sequence of zero length?

deshipu, 2016-01-16:
Let's seem if you repeat something of ``a`` length ``b`` times, you get something of length ``a``×``b``. So repeating something of length 0 should give you a sequence of length 0×``b``, which is 0. You could try arguing that you are repeating it infinite number of times, in which case ``b`` is infinity and the whole thing doesn't make sense, but that's not true. In reality, you are just repeating it ``b`` times, but you don't know ``b`` until the repeating is interrupted. So ``b`` is unbound but finite, which makes 0×``b`` zero, which means that repeating an empty sequence should yield an empty sequence. QED :-)

markshannon, 2016-01-16:
Which is what it does. The iterator over an empty sequence starts exhausted.

deshipu, 2016-01-16:
Great, we have it working for one case. Now let's make it work for the non-trivial cases.

markshannon, 2016-01-16:
It does. You pass it an iterable, and it iterates over it an infinite number of times, stopping should the iterable become empty.
I freely admit that it is not as nice an interface as `itertools.cycle`, but you know how little RAM we have.

deshipu, 2016-01-16:
Maybe it should be called ``maybe_repeat()`` and the param renamed to ``perhaps_loop`` then?

markshannon, 2016-01-16:
Well, according to your earlier logic, it always repeats. It just doesn't crash if the iterable that becomes empty.

deshipu, 2016-01-17:
It doesn't, and you know it, and this sophism does nothing to reduce the surprise of the users when they encounter this behavior, or the inconsistency of this behavior. Just think about how the documentation of the ``loop`` parameter would need to look like:

If ``loop=True``, the function might repeat the sequence indefinitely, but only sometimes, depending on the type of the parameter. Lists are mostly fine, unless they have been reversed. Tuples are mostly fine. Iterator generator functions are right out, unless they are infinite already. Don't even ask for user-defined types.

markshannon, 2016-01-17:
Or you define it formally as I have done.
What interface do you propose?

deshipu, 2016-01-17:
I think the correct behavior would be to either:

1. Remember the old values, like the ``itertools.cycle`` does, with possible special case for types that support iterating over them multiple times, to save memory.
2. Raise an exception if it can't be repeated without remembering the values, explaning the situation.

markshannon, 2016-01-17:
1. Is impossible due to memory constraints.
2. You cannot, in general, tell if an iterable can be repeated or not.
What we could do is raise an exception should the iterable become empty.
That way it remains as general as possible, but you will get an error if you pass in an exhaustable iterator and set `loop=True`.

deshipu, 2016-01-17:
> You cannot, in general, tell if an iterable can be repeated or not. What we could do is raise an exception should the iterable become empty. That way it remains as general as possible, but you will get an error if you pass in an exhaustable iterator and set ``loop=True``.

This is what I meant. What do you think?

markshannon, 2016-01-17:
Yes, seems sensible. 
After all, this is going to be very rare and only likely to happen by accident or to a sophisticated user.
If it is an accident, then raising an exception is the best.
If it is a sophisticated user, then they'll find another way to do what they want.
So,
```python
def repeat(iterable):
    while True:
        iterator = iter(iterable)
        try:
            item = next(iterator)
        except StopIteration:
            raise ValueError("Cannot repeat empty sequence or exhausted iterator")
        yield item
        yield from iterator
```

deshipu, 2016-01-17:
Now the question is what to do with the async version, when by the time you reach the exception the program is happily executing some completely different code...

ntoll, 2016-01-17:
Oh lordy lord. This one is for @markshannon and @dpgeorge:

https://www.youtube.com/watch?v=kcHKm0cm-jI

Personally I think @markshannon has the right outlook with: 

"After all, this is going to be very rare and only likely to happen by accident or to a sophisticated user.
If it is an accident, then raising an exception is the best.
If it is a sophisticated user, then they'll find another way to do what they want."

While I understand and sympathise with @deshipu's point about consistency with upstream (grown-up Python) modules I believe this isn't too onerous a problem. Kids are used to simplified or "special" educational cases from which they are expected to graduate.