bbcmicrobit/micropython #156: ioctl error from pyboard.py when running in a Docker container

ncoghlan, 2016-01-17:
I have @ntoll's microrepl project successfully communicating with the micro:bit from an Ubuntu container running on a Fedora host [1], but attempting to run pyboard in the same container is failing with an ioctl error:

    $ python3 tools/pyboard.py examples/conway.py 
    Traceback (most recent call last):
      File "/home/microbitdev/.local/lib/python3.4/site-packages/serial/serialposix.py", line 322, in _reconfigurePort
        orig_attr = termios.tcgetattr(self.fd)
    termios.error: (25, 'Inappropriate ioctl for device')

    During handling of the above exception, another exception occurred:

    Traceback (most recent call last):
      File "tools/pyboard.py", line 293, in <module>
        main()
      File "tools/pyboard.py", line 280, in main
        pyb = Pyboard(args.device[0], args.baudrate, args.user, args.password)
      File "tools/pyboard.py", line 125, in __init__
        self.serial = serial.Serial(device, baudrate=baudrate, interCharTimeout=1)
      File "/home/microbitdev/.local/lib/python3.4/site-packages/serial/serialutil.py", line 282, in __init__
        self.open()
      File "/home/microbitdev/.local/lib/python3.4/site-packages/serial/serialposix.py", line 296, in open
        self._reconfigurePort()
      File "/home/microbitdev/.local/lib/python3.4/site-packages/serial/serialposix.py", line 325, in _reconfigurePort
        raise SerialException("Could not configure port: %s" % msg)
    serial.serialutil.SerialException: Could not configure port: (25, 'Inappropriate ioctl for device')

Looking at the code for the respective utilities, microrepl is using the higher level serial.tools.miniterm API to set up an interactive console, while pyboard is using serial.Serial directly.

[1] https://github.com/ncoghlan/microbit/tree/master/devenv

dpgeorge, 2016-01-18:
When using pyboard.py you need to specify the device first, then the script to run, eg:
```
$ tools/pyboard.py /dev/ttyACM0 examples/conway.py
```
It doesn't do any fancy autodetection of the device :)

ncoghlan, 2016-01-18:
Oops - and right after I sent @ntoll the world tour post saying I hadn't got it to work! :)

ncoghlan, 2016-01-18:
Now that I know what I was doing wrong, I can see exactly what the error was trying to tell me (i.e. the file descriptor for the source file isn't a valid device FD).

However, it may be worth statting the first argument explicitly and reporting a less cryptic error message if it isn't a device

dpgeorge, 2016-01-18:
> However, it may be worth statting the first argument explicitly and reporting a less cryptic error message if it isn't a device

Well, at the very least it should catch any IOError or OSError and just say "device not found".  Then you know that the first argument should be the device.

dpgeorge, 2016-04-09:
Error message is now much nicer when device not found.

closed 2016-04-09 by dpgeorge