bbcmicrobit/micropython #157: Problem with stopping ongoing background display.animate

dpgeorge, 2016-01-18:
The following will cause the device to lock-up:
```python
>>> display.animate('12345', 100, stride=1, wait=False, loop=True)
>>> display.animate('12345', 100, stride=1)
```
This might be fixed when the display async stuff is finalised, #153, #154.

markshannon, 2016-01-21:
Testing #154, there is no `animate` any more, but the following works:
```python
>>> display.show('12345', 100, wait=False, loop=True)
>>> display.show('12345', 100)                       
```
as does:
```python
>>> display.scroll('12345', 100, wait=False, loop=True) 
>>> display.scroll('12345', 100)                       
```

ntoll, 2016-01-21:
I see the same as @markshannon

closed 2016-01-21 by ntoll