bbcmicrobit/micropython #166: compass.calibrate() bricks the device

ntoll, 2016-01-22:
In a build from current master, at the REPL, try:

```
>>> compass.calibrate()
```

The device freezes until you press reset.

dpgeorge, 2016-01-22:
It's because of the recent update to the latest version of the DAL.  It was something I was aware might happen.  Needs investigation.

ntoll, 2016-01-22:
While this doesn't solve the problem, I notice that we call the `calibrateAsync` function which will be removed. We should just call `calibrate` instead (which is essentially all that happens at the moment).

whaleygeek, 2016-01-22:
All my TD scripts that use compass broke overnight. Cal is auto now I
think, as is tilt compensation. Looks like change made in a non backwards
compatible way

markshannon, 2016-01-24:
https://github.com/lancaster-university/microbit-dal/issues/82

markshannon, 2016-01-24:
Looking at the [data sheet](http://www.nxp.com/products/sensors/magnetometers/sample-data-sets-for-inertial-and-magnetic-sensors/freescale-high-accuracy-3d-magnetometer:MAG3110?fpsp=1&tab=Documentation_Tab), it appears that the magnetometer is quite accurate.
However, it can only measure the actual magnetic field, not the Earth's magnetic field, which is distorted by all the conducting material on the micro:bit board.
Given all micro:bit boards will be the same, can we not just calculate the correction factors once and then there would be no need to calibrate them?
@dpgeorge You studied physics rather more recently than me, what do you think?

carlosperate, 2016-01-24:
Wouldn't all the addon boards being created to be placed very close in parallel with the microbit also affect this measurement?

ntoll, 2016-01-25:
Given @finneyj's reply here: https://github.com/lancaster-university/microbit-dal/issues/82 we should work out next steps. I quite like @dpgeorge's suggestion in the comments on the DAL issue: "Can we enable the MicroBitDisplay object before calling the calibration, and then disable it once it's done?"

Given that the DAL's `calibrate` is blocking then the display won't (or perhaps shouldn't?) be doing anything to make this a serious problem. Or am I missing something here..?

dpgeorge, 2016-01-25:
I tried the simplest thing, to enable the DAL display, do the calibration, then disable the DAL display.  And it almost works!  It looks like this:
```c
mp_obj_t microbit_compass_calibrate(mp_obj_t self_in) {
     microbit_compass_obj_t *self = (microbit_compass_obj_t*)self_in;
+    uBit.systemTicker.attach(&uBit, &MicroBit::systemTick, MICROBIT_DISPLAY_REFRESH_PERIOD);
+    uBit.display.enable();
     self->compass->calibrateAsync();
+    uBit.display.disable();
+    uBit.systemTicker.attach(ticker, MICROBIT_DISPLAY_REFRESH_PERIOD);
     return mp_const_none;
 }
```
The bit that doesn't work is that the calibration function gives an out-of-memory error when doing the matrix transposition (because it allocates on the heap for the new, transposed matrix).  @finneyj did you see this behaviour before?

ntoll, 2016-01-26:
Great stuff! Just remember that we should be calling `self->compass->calibrate()`. ;-)

Not so great that we get an out-of-memory error. :-/

markshannon, 2016-01-26:
@whaleygeek Have you reported your TD script breakages to https://github.com/lancaster-university/microbit-dal?

dpgeorge, 2016-01-26:
I have the calibration working fine under uPy, but adoption is now pending a more serious issue: https://github.com/lancaster-university/microbit-dal/issues/85

dpgeorge, 2016-01-26:
> Given all micro:bit boards will be the same, can we not just calculate the correction factors once and then there would be no need to calibrate them?

@markshannon I had a quick look at how consistent the calibration data is and it's not a good outcome: even using the same microbit, in the same location, you get pretty different results for the values, eg:
```
CAL(-255432, 209004, 353981)
CAL(-264186, 193946, 109738)
CAL(-256946, 249507, 160277)
CAL(-197916, 321708, 12046)
CAL(-265891, 258603, 86923)
```
They are the x,y,z average values used for the calibration.  @jamesadevine also did some testing and got these numbers:
```
CAL(-155605, 174637, 445991)
CAL(66707, 336171, 886896)
CAL(-418155, 557441, 269014)
CAL(-344345, -12279, 323963)
```
James said that he and Joe did some testing already in this area and found that the calibration was not something you could fix to a certain value.

blmorris, 2016-01-26:
My $0.02 based on some IMU hardware work I've done with MicroPython: the problem isn't even as simple as calibrating once for a given board and keeping that value constant for the life of a given board.
If there is any ferromagnetic material on the board (and there is, lurking in all kinds of unexpected places: SMT capacitors, IC chip leads, connectors, EMI shielding, and of course any ferrite inductors or chokes) then they will carry their own magnetic field that will influence the compass. These local fields can be compensated for with compass calibration; however, these materials will be re-magnetized if a board is ever exposed to a sufficiently strong magnetic field. At that point any old calibration values will be invalid and need to be measured again.

dpgeorge, 2016-01-26:
Thanks @blmorris!  Very insightful and useful info.

dpgeorge, 2016-01-26:
The problem with excessive memory usage should be fixed very soon (the test branch of the DAL with the fix seems to work).

So, how shall we deal with getting calibrate working with uPy?  Due to time constraints (ie wanting v1.0 yesterday) I'd say we go with a temporary solution for now that just works.  This can be simply enabling the display and uBit system ticker during calibration, and then disabling it when done.  I've testing this and it works, see the patch above.

In the long term we can have a more comprehensive solution.  But that requires coming up with a way to do calibration (do we want a game? somehow the user needs to know in what way to tilt/rotate the device), testing it, implementing and documenting it.  And our way needs to be supported by the DAL, so there will be some back-and-forth getting both sides in sync.

ntoll, 2016-01-27:
Agree with the immediate solution. Given the upstream DAL want to refactor to make this stuff easy for client languages like MicroPython I think we should engage with @finneyj et al once constraints are removed.

ntoll, 2016-01-30:
Fixed by #173

closed 2016-01-30 by ntoll