bbcmicrobit/micropython #172: Anyone interested in porto the microbit module to python 3.3

chickengirl1976, 2016-01-28:


ntoll, 2016-01-28:
Not quite sure what you mean. MicroPython *is* a flavour of Python 3.

dpgeorge, 2016-01-28:
Maybe @chickengirl1976 means to make microbit.py run under CPython?  So you can do prototyping on a PC.

chickengirl1976, 2016-01-28:
As a teacher in secondary school we already use Python 3.3 and it would be great if we could just import the microbit module rather than installing yet more software or using a browser to access the software. 

(Hi Nicholas = Joe Finney here!) I think the question here is can you "remote control" a micro:bit running micropython from a python environment running on an attached PC. :-)

dpgeorge, 2016-01-28:
> I think the question here is can you "remote control" a micro:bit running micropython from a python environment running on an attached PC.

Yes, already possible using tools/pyboard.py.  This allows to run scripts from the PC directly on the microbit.  But probably you want to take this a step further for true "remote control" behaviour.  Can you provide use-case example(s) so we understand exactly what is being requested?  Thx!

ntoll, 2016-01-28:
Yes... use Mu see https://github.com/ntoll/mu it has a built in REPL client as well as a code editor. FEEDBACK WELCOME!

(You'll need to install the ARM drivers for USB/serial, but look in the README for Mu for instructions)

ntoll, 2016-01-28:
BTW Mu will work on Windows, OSX, Linux and RaspberryPi. RPi Foundation are making it part of Raspbian and it'll launch when you plug your micro:bit into a Pi. ;-)

chickengirl1976, 2016-01-28:
Mu is great .... I was thinking about school's network manager being reluctant to install yet more software. Just having it available in the current python library as we use Python in Year 10/11 would be the ideal situation. 

We would like the microbit just as a useable piece of hardware for other year groups too. Would be great to just drag and drop .py files onto the Microbit.

chickengirl1976, 2016-01-28:
BTW we have Rasp Pi but not enough for a class group - although they ideal - we need a whole class to have hands on

ntoll, 2016-01-28:
You don't need to install anything... just double click the `.exe` and it'll work. It has a "flash" button that'll autodetect the micro:bit and push the code onto the device. No drag and drop needed.  You only need to install the drivers if you want the REPL (which is a really cool feature). 

Also, we're working with some school network managers - they're going to package it all up in the correct way so *any* network manager can just deploy them across the whole network. :-)

Keep the feedback coming..! It's invaluable. Happy to help in any way that we can.

chickengirl1976, 2016-01-28:
Sounds good - I'll give it a go at school. 

Hope the network doesnt just stop it being installed as its an exe file

Fingers crossed. Thanks

ntoll, 2016-01-28:
Hmm... as a teacher, you don't have access to install the software you need..? That's a really bad situation. Alternatively, I can chivvy up our captive network managers and ping you with a package you could test on your captive network manager. ;-)

chickengirl1976, 2016-01-28:
Very few teachers can install software on the network. Most school don't even have Python installed on their networks due to security issues. 

Anything you can send me to test would be great!

whaleygeek, 2016-01-28:
I *think* @chickengirl1976 is talking about remote control in the sense
that I demonstrated to you @ntoll just before PyConUK - i.e. write python
code on the PC in their existing Python environment, that uses the
micro:bit as a remote input and output device.

The way I did this was to create a microbit module that you import on the
PC like this

import microbit

And then for each micro:bit feature, the PC version would send the
appropriate REPL to MicroPython running on the micro:bit, it would 'do it's
stuff' and return the result.

So, writing this on the PC

import microbit

while True:
  x = microbit.accelerometer.get_x()
  y = microbit.accelerometer.get_y()
  print("pos %s %s" % (x, y))
  sleep(1)

The bulk of this loop runs on the existing installed python on the PC. When
it calls microbit.accelerometer.get_x() it sends a message to the micro:bit
REPL to get the x acceleration, it returns the value, and that is assigned
to the x variable in the python running on the PC.

So, the micro:bit just becomes a sort of remote sensing device.

microbit.display.scroll("hello world")

Running this on the PC would scroll Hello World on the micro:bit display.

The micero:bit MicroPython API has changed quite a lot since I first wrote
this, but if this is what you are talking about, I could upload the code to
github and someone else could help me update it to the latest API :)

David

chickengirl1976, 2016-01-28:
David - that is exactly what I was thinking! sorry it wasnt clearer....

More than happy to test the Mu too

ntoll, 2016-01-28:
@whaleygeek haven't you just described a REPL..? :-P

(Yeah, I see what you mean though)

ntoll, 2016-01-28:
@chickengirl1976 @whaleygeek aha... thanks for the clarification. :-)

chickengirl1976, 2016-01-28:
would need the drivers for the REPL - and thats a software installation issue again .....

ntoll, 2016-01-28:
@chickengirl1976 it's not an installation issue, it's a network manager issue. :smile: IIWY I'd hire a helpful network manager and make it part of their job description to install the tools you request. How hard can it be? :-)

chickengirl1976, 2016-01-28:
Have you ever met a network manager in a school ? - to be fair ours is great but other schools aren't so lucky!

Still not straight forward to get it installed

ntoll, 2016-01-28:
I'm pulling your leg and quite understand. I used to be a senior teacher in a secondary school (I was a head of music). Perhaps something to raise with your SMT so your network manager is able to act unencumbered.

I've met many troll-like examples of the "network manager" species which is why we're trying to make it as easy as possible for them to install it via a ready made package.

The alternative is, of course, to use a Pi, in which case, with the next version of Raspbian, it'll come with Mu installed and ready set up.

:-)

Keep the feedback coming!

whaleygeek, 2016-01-28:
Here is where I got to with it, but it's very old, the API has changed quite a lot since.

[microbit.py.zip](https://github.com/bbcmicrobit/micropython/files/108411/microbit.py.zip)

It uses pyserial, which I have not included, but if you put the standard pyserial distribution in a sub folder called 'serial' it sorts it's life out and you don't have to install pyserial as a dist package, it works embedded inside the folder. So copying a single folder into the student workspace can make this work.

Also, the portscan tool I wrote "auto locates" the correct serial port number on mac, pc, linux, raspberry pi.
[portscan.zip](https://github.com/bbcmicrobit/micropython/files/108417/portscan.zip)

All of this is based on the work I did to build an Arduino based GPIO adaptor for PC/MAC so that I could support electronics in my python coding book - the original source for this is in the anyio package here:

https://github.com/whaleygeek/anyio

I'm really swamped with lots of stuff at the moment, but if someone wanted to take the lead on this and help me push this into shape as per the current micro:bit MicroPython API, I'd be willing to steer the activity a bit.

I did think that the method I used to translate from API methods to strings sent over the REPL could use some meta-love (i.e. I think a lot of it could be auto generated).

ntoll, 2016-01-28:
@whaleygeek great stuff... it works in the same way as urepl... (PySerial). Is this work in a repos somewhere so we can fork/collaborate..?

whaleygeek, 2016-01-28:
If there is a problem installing serial drivers, we need to raise this with BBC/ARM - the whole mbed experience is *supposed* to be zero install, specifically for schools that have poor or no IT support.

whaleygeek, 2016-01-28:
@ntoll it's not in a repo, but it *could be* and then you could fork it from there. It doesn't actually work at the moment due to changes in the API, but if I put it in a repo 'as is', would that be the quickest way for others to jump in and help with it?

ntoll, 2016-01-28:
It's been raised with Microsoft already. The problem is Windows. I hear ARM is all over it and pushing hard for MS to fix the problem.

Yes, stick it in a repos, I'll take a look later..! :-)

whaleygeek, 2016-01-28:
Creating a repo now. Will send you the link to it when it is ready, and people can fork it and help to give it some extra love and turn it into something we can use. Ta.

ntoll, 2016-01-28:
@whaleygeek as always, many many thanks! :smile:

chickengirl1976, 2016-01-28:
Looking forward to the development of this!!

remay, 2016-01-28:
I've got a project underway to implement a serial control interface to a micro:bit.  The intention initially is to closely match what the existing bluetooth API supports - mainly as I want to be able to use a serially controlled tehered micro:bit to access the bluetooth APIs on other micro:bits, allowing a more IoT experience without having to have a mobile phone.

I've got rather bogged down in keeping up with the BLE security 'enhancements' that the BBC has forced on the project, but did before Christmas have an unsecured version running (and integrated with Node-Red).  I'm also being seriously hampered by the fact that I need the S130 BLE softdevice to run the tethered micro:bit as a BLE central device, and it's size, along with the micro:bit runtime requirements mean that I have only 512 bytes of stack to play with.

A python module that integrates with this might also be an interesting idea?

Currently (non-working) code is in my microbit-ble-bridge repository.   And some initial thoughts on the final API here: https://gist.github.com/remay/fb94d3cc40fc5dc802f4

remay, 2016-01-28:
If MS/BBC/ARM could solve the serial drive install issue on Windows that would be fantastic.   It 'just works' on my mac.

dpgeorge, 2016-01-28:
> If MS/BBC/ARM could solve the serial drive install issue on Windows that would be fantastic. It 'just works' on my mac.

+1e100

It also "just works" on Linux.  I have pushed MS from a few directions on this, but I think it's a lost cause :(  It would help so many projects, not just micro:bit.

whaleygeek, 2016-01-28:
Ok, this is now in a repo here:

https://github.com/whaleygeek/mb_remote

It won't work with the latest API, but at least it is in a place that others can fork it and give it some love, and hopefully it will turn into something of use to all.

sparkslabs, 2016-01-28:
As background, @chickengirl1976 asked me this via DM on twitter, and my response was: (lightly editted)

- If you were to do so, building on top of ntoll's "mu" would be a good starting point, since that already has the bulk of the things you'd "need" from a library pespective, and is already packaged up/usable
- It's also a very logical place to use it
-  Though, again, I can see real benefits in it being "just" a module you import and use - say based on pygame/tk/qt/etc for the display
- Having a very silly thought about using a wiimote to control a virtual microbit - so you can simulate shaking etc, and the button presses :)

ie the ability to effectively have a microbit simulator for CPython which allows you to run/test microbit micropython programs without using a device. That's not quite the same as what's described above. (Though I can see both as being useful in different way)

DavidWhaleMEF, 2018-12-13:
It looks like this ticket is finished now (logged in 2016 and no activity for about 2 years?)

For completeness, the work I was discussing with @ntoll above surfaced as bitio (I gave a talk about it at the 2017 PyconUK), it probably answers a number of the questions and issues discussed in this ticket.

https://github.com/whaleygeek/bitio

Can we close this ticket now, do you think @MsRussellMCHS ?