bbcmicrobit/micropython #174: Happy to sacrifice some FLASH for persistent configuration data?

finneyj, 2016-01-29:
Folks,

I've been asked through several avenues to enable persistence of configuration / metadata that would across reboots (e.g. compass calibration data, temperature calibration data, BLE connection / bonding metadata, etc.

Under Bluetooth enabled builds there's an area reserved for doing this - the UICR. This is a 128 byte region that lurks at memory address 0x10001000.  I we just started using this in micro-bit DAL it would most likely stomp all over some MicroPython runtime code, so we obviously don't want that...

Can I solicit feedback on:

1) do we want to see calibration data persisted (all user feedback i've seen thus far wants this)
2) do we want to block out this 128 byte region UICR region for MicroPython builds also
3) would we rather block out out own (shared) configuration area in a 1K flash page somewhere?

My personal feeling is that we do want to persist such data, and that the final option (a 1K page somehwhere) might be best, as it could also conceivable support other modules that may wish to use this going forward...

dpgeorge, 2016-01-29:
The flash starts at 0x00000000 and goes up to 0x00040000, so 0x10001000 is way out of that range... or am I missing something?

Also, was there talk of making some persistent data storage API as well (eg to store temperature readings over time)?  That will need more than 1k.

deshipu, 2016-01-29:
Just one remark. If we are going to have persistent configuration, I think it's very important to make sure that:
1. it can be easily reset to factory defaults, and
2. it's not possible to get it into a state in which the board doesn't work anymore.
Without the storage, fixing the board has been easy -- just flash new firmware and reset it. Persistent configuration adds another item to worry about when debugging problems.

dpgeorge, 2016-01-29:
One can easily make a blank.hex file that erases the entire flash.  That would reset to a default.  We could provide such a file for users in case they have problems.

finneyj, 2016-01-29:
Agree it's super important that we ensure the device can't be bricked. A USB flash will erase everything on the device, so will indeed return to a known state. I think there's separate discussion to be had about if/how we want to maintain state across re-flashes to have a little way down the line. For now, I was just thinking about persisting across reboots, which is a safer first step.

@dpgeorge Apologies, i think the Nordic data sheet assumes a base offset for physical addresses. it would map to 0x00001000. 

A more scalable data storage API would be a very valuable addition to the project I think. I started drafting up a lightweight, general purpose implementation of the mbed FileSystem interface that would run inside program memory ages ago that we might use for data logging apps etc (and potentially even for .py files for you guys which would be awesome), but keeping up with the firehose of issues has meant I've had to refocus on more mundane things. It would be good to resurface this as an open project for folks to work on soon.

I was thinking of this configuration block as independent of a file system though - something simpler (probably a table of name/value pairs or the like). I was suggesting 1K as that's the FLASH page size on the nrf51822 so keeps things simpler...

dpgeorge, 2016-01-29:
0x00001000 is going to clash with everyone, not just us.  How will TD deal with it?  Simplest thing is to just start the .hex file at 0x00001400.  Then you could put 4 * 1k pages starting at 0x00000000, and also the nordig 128 bytes at the page at 0x00001000.

finneyj, 2016-01-29:
@dpgeorge Apologies - I wasn't clear here. The UICR is a reserved area in the big Soft Device blob that forms the majority of the BLE stack, so is already reserved in most builds (all those using bbc-microbit-classic-gcc as a target). this includes TD, JS, Block, etc. I don't want to use it though as it will clobber you guys badly, as your using the bbc-microbit-classic-gcc-nosd target (which doesn't include Soft Device).

I don't think we can reserve the start of memory I'm afraid, as SoftDevice is a monolithic blob that assumes it's resident at the start of memory.

If you're happy to let a FLASH page be dedicated to this, I'd suggest something toward the top end of memory, near with the BLE bootloader lives on the bbc-microbit-classic-gcc build.

dpgeorge, 2016-01-29:
I see.  Ok, then a 1k page at the top of flash sounds like the best option.

Except: we store Python scripts there.... specifically, we put them starting at 0x3e000, which gives us 8k of space (which is actually pretty tight, I'd like to increase it a bit now that we know how big the MP firmware is).

If we want to have room in the future to have a "filesystem" I think we should design something now which takes that into account.

For example, we could dedicate the top 24k to data.  The top 1k could be persistent config.  Then there could be 23k for a filesystem, of which we use some for storing the Python scripts.

remay, 2016-01-29:
@finneyj The nrf51 Reference Manual suggests that UICR is available regardless of whether there is a soft device present or not, and is accessed via the registers at and above base address 0x10001000.   It doesn't appear to say where the data is actually stored, but at least hints that is separate from the code flash area.  (and [this page](https://devzone.nordicsemi.com/documentation/nrf51/4.4.1/html/group__uicr__config__example.html) hints that there might be actual storage mapped to 0x1000100+offsets, as if appears that it can be programmed by flashing to these addresses.

The problem with UICR is that there are only 128 bytes of usable memory.

I'd be in favour of extending the area of flash just below the bootloader that we already use for BLE bonding information storage - with some reserved for the DAL and some reserved for end user programs to access.

finneyj, 2016-01-29:
That might track. For BLE enabled builds there's a bootloader that lives up at the top of memory also - 16K, plus another 2K for security information). We'd need to avoid this in both builds, so having a cross language "RESERVED" space in the top 24K seems pretty good to me.

I'm sure we'd make the file system API configurable as to where it lived, and optional of course.

Sounds like a plan...

finneyj, 2016-01-29:
@remay - that's very interesting! Wonder where on earth the UICR goes then?

remay, 2016-01-29:
> having a cross language "RESERVED" space in the top 24K seems pretty good to me

+1

markshannon, 2016-04-06:
Flashing the device erases *all* the memory, the hardware is designed that way. Data can only persist until the device is flashed. 
Given that, it doesn't matter where in memory you choose to put the data, just declare a symbol and use the address that the compiler allocates.
The UICR, it is only 128 and gets erased the same as all other flash.

closed 2016-04-06 by markshannon

dpgeorge, 2016-04-07:
> Flashing the device erases all the memory, the hardware is designed that way.

You can erase single pages.  I think it's the Freescale support chip that's programmed to erase the entire flash (and I guess it could be changed, but probably won't be).

So, erasing the whole flash on download of new firmware does simplify things because you can change the layout of the filesystem from one version to the next.  But it still doesn't solve the original issue of this issue, namely: where/how should the DAL allocate persistent data so that it doesn't clash with uPy.

remay, 2016-04-07:
The flash can be partially overwritten.   The Nordic loader has support for BLE flashing which can do this (it is capable of leaving the softdevice and nordic loader unchanged at the bottom and top of flash respectively).   The Freescale also has support for this - and I believe it would be possible to write something that did this today using the CMSIS-DAP protocol (as used by PyOCD: https://github.com/mbedmicro/pyOCD).

The software behind the USB Mass storage emulator on the Freescale that is used to flash the Nordic does not have capability to do anything other re-write the whole flash bank.  I have heard a rumour (from what I believe is a reliable source) that someone is looking at new firmware for the Freescale that can just change executable part of the flash.  I don't have any more details though.