bbcmicrobit/micropython #182: String.format() right justification not working for string arguments

realcundo, 2016-02-01:
When playing with Marja (using [`firmware.hex`](https://github.com/ntoll/microrepl/blob/master/firmware.hex)) I came across this:
```python
print("x{:>4}x".format(42))
print("x{:>4}x".format("42")
```
produced
```
x  42x
x42  x
```
Not a big deal but I was wondering whether this was intentional, consequence of squeezing as much Python into as little space as possible (justification is ignored?), or a genuine bug.

dpgeorge, 2016-02-01:
This is now fixed in the upstream code.  I'll see if I have time to sneak those fixes into microbit before v1.0....

ntoll, 2016-02-01:
Yes please :-)

dpgeorge, 2016-02-01:
Actually, this is already fixed.  You must have an older version of the firmware.

closed 2016-02-01 by dpgeorge