bbcmicrobit/micropython #193: No LICENSE file?

gbaman, 2016-02-06:
I just noticed, the MicroPython repository has no LICENSE file. Without it, the appropriate license has to be specified in each file. Which in nearly all (if not all) the code files, it has been done (with MIT). But that doesn't cover stuff like /tools, /docs (what I was interested in) or /examples so technically, they are covered under "No License", meaning in general ["nobody else may reproduce, distribute, or create derivative works"](http://choosealicense.com/no-license/) from these sections.

Is there any reason a LICENSE file hasn't been added? Am I missing something?

ntoll, 2016-02-06:
Good spot. We should add one.

zitterbewegung, 2016-02-08:
If we are adding a license do we have to contact everyone that contributed
code for consent to release it under a license? I don't remember that there
is a contributor agreement either .

JoeGlancy, 2016-02-09:
What about examples? Quite a few of them have no licensed or terms attached. Perhaps public domain would be ideal for examples? But then the respective authors would need to be contacted.

dpgeorge, 2016-02-09:
I think that, for the C/C++ code at least, it was implicit that it's under the original license of MicroPython, ie MIT.  But I agree we need to add a LICENSE file and make sure people are happy with it.

dpgeorge, 2016-02-09:
Dear all contributors to this repository: @dpgeorge @ntoll @markshannon @deshipu @matthewelse @willingc @tomviner @alanmjackson @funkyHat @paulegan @jrmhaig @alexwlchan @ncoghlan @gbaman @kwadrat @JoeGlancy 

Are you ok with your code being licensed under the MIT license?  If you're not familiar with this license, see the links below (it's very easy to understand, compared with licenses in general).

https://opensource.org/licenses/MIT
https://en.wikipedia.org/wiki/MIT_License

Thanks!

ntoll, 2016-02-09:
Completely happy.

I presume that if code is explicitly licensed as something else this will remain in place (the only code I can think of are some examples Larry and I wrote and donated to the public domain).

jrmhaig, 2016-02-09:
Yes, I'm happy with this.

dpgeorge, 2016-02-09:
> I presume that if code is explicitly licensed as something else this will remain in place

Yes, individual files can be licensed differently to the overall license.

JoeGlancy, 2016-02-09:
Go for it

gbaman, 2016-02-09:
Yip, I am happy with it.

deshipu, 2016-02-09:
I agree

kwadrat, 2016-02-09:
Yes, I am OK with my code being licensed under the MIT license.

ncoghlan, 2016-02-09:
Good catch, and yes, I am fine with my contributions being published under the MIT license.

markshannon, 2016-02-09:
Yes, for all the code I have contributed so far.

For all l future contributions: yes, unless I explicitly state otherwise.

funkyHat, 2016-02-09:
Yes I'm happy with MIT :)

alexwlchan, 2016-02-09:
MIT license is fine for me. :-)

Neon22, 2016-02-09:
yes. MIT 
    mine are labelled public domain - should we edit ?
    
    On 2/9/2016 11:37 PM, Damien George
      wrote:
    
    
      Dear all contributors to this repository: @dpgeorge @ntoll
        @markshannon @deshipu @matthewelse
        @willingc @tomviner
        @alanmjackson @funkyHat @paulegan
        @jrmhaig @alexwlchan
        @nco ghlan @gbaman @kwadrat
        @JoeGlancy 
      Are you ok with your code being licensed under the MIT license?
        If you're not familiar with this license, see the links below
        (it's very easy to understand, compared with licenses in
        general).
      https://opensource.org/licenses/MIT
        https://en.wikipedia.org/wiki/MIT_License
      Thanks!
      —
          it on GitHub.
      
        
          
          
        
        
      
      No virus
        found in this message.
        Checked by AVG - www.avg.com
        Version: 2016.0.7357 / Virus Database: 4522/11587 - Release
        Date: 02/08/16

dpgeorge, 2016-02-09:
@Neon22 if it's already public domain, no problem.

willingc, 2016-02-09:
Yes, I am fine with MIT :-)

tomviner, 2016-02-09:
I approve this message. Happy with MIT.

ntoll, 2016-02-09:
@tomviner https://www.youtube.com/watch?v=rCcqMzuezaI ;-)

zitterbewegung, 2016-02-11:
I have no issue with this.

dpgeorge, 2016-02-11:
Ok, just waiting on responses from: @matthewelse @alanmjackson @paulegan

paulegan, 2016-02-11:
Yes to MIT

matthewelse, 2016-02-11:
MIT's fine with me!

dpgeorge, 2016-04-09:
Addressed by #229.

closed 2016-04-09 by dpgeorge