bbcmicrobit/micropython #196: External buttons

gbaman, 2016-02-06:
Any chance of having some way to set up external buttons? I know you could just use the read_digital() function, but the A and B buttons have a number of rather nice additional features like ```was_pressed()``` and ```get_presses()``` which would be nice to have for external buttons if there was a way to set up additional pins as buttons.

dpgeorge, 2016-02-06:
I agree that it would be nice to make the button API more general, and extend it to pins.  Really it should be the other way around: the Pin API should have these features (was_high, was_low) and buttons should just be a special case of the Pin API (and use the same code).

This would require modifications to the DAL (or support for debouncing etc on our side, probably a better way to go...).

JoeGlancy, 2016-02-10:
On this topic, I'm attempting to make some kind of headway into assigning a MicroBitButton instance to a certain pin, however: the memory for the object is dynamically allocated using microbit_malloc, including the (MicroBitButton) class instance itself. Is there any way to call its constructors and deconstructors, or would it be better to move the body of the MicroBitButton constructor and deconstructor to a separate routine which can be called externally?

Also, a microbit_button_obj_t can be cast and returned as an mp_obj_t for the Python API, right?

dpgeorge, 2016-02-10:
> Is there any way to call its constructors and deconstructors,

Well, using new and delete :)

> or would it be better to move the body of the MicroBitButton constructor and deconstructor to a separate routine which can be called externally?

That would probably be better, to make the DAL more C friendly and module.

> Also, a microbit_button_obj_t can be cast and returned as an mp_obj_t for the Python API, right?

Yes.

JoeGlancy, 2016-02-11:
okey dokey, thanks!