bbcmicrobit/micropython #197: Nicer error messages for (syntax?) errors

ntoll, 2016-02-07:
I've been sent this buggy script by a partner. When they flash the device they just see "L????????????" scroll across the display. I suspect could probably do better than this - especially given the script is full of syntax errors.

```
from microbit import *

def random_light(intensity):    

    #lights an LED in first, second, fourth or fifth columns

    xdim = random(5)

    ydim = random(5)

    if xdim = 2:

        random_light()

        else:

           sleep(random(10000)) 

           display.set_pixel(xdim, ydim, intensity)

           if xdim < 2:

               return "left"

               else:

                   return "right"

    

def button_input():

    #prevents cheating by pressing both buttons

    while True:

    if microbit.button_a.is_pressed() and microbit.button_b.is_pressed():

        button_input()

    elif microbit.button_a.is_pressed():

        return "left"

        break

    elif microbit.button_b.is_pressed():

        return "right"

        break

def start():

    #sets up the display

    display.scroll("are you ready?", delay=100)
    for x in xrange(0,5):
        for y in yrange(0,5):
            display.set_pixel(x, y, 9)
            sleep(5)

#keeps track of current score using a stack

init_score = [0]

   

def cur_score():

    #displays last score in stack

    return init_score[-1]

def affect_score(amount):

    #adds new score to stack

    init_score.append(cur_score() + amount)

    return cur_score()

start()

while True:

    # game loop - increments brightness of LED as score increases affecting contrast against surrounding LEDs

    if cur_score() < 10:

        this_turn = random_light(9 - cur_score())

        #compares button with LED LH or RH location

        if this_turn == button_input():

            affect_score(1)

            Image.HAPPY

            sleep(1000)

            display.scroll("Great! score = " + cur_score(), delay=100)

        else:

            Image.SAD

            sleep(1000)

            display.scroll("game over! score = " + cur_score(), delay=100)

            microbit.reset()

    else:

        display.scroll("You win!", delay=100)

        microbit.reset()

```

ntoll, 2016-02-07:
Actually, this script reproduces the effect with the latest build of MicroPython...

```
# There is no longer a random function.
from microbit import *

display.scroll(str(random(10)))
```

markshannon, 2016-02-07:
This is a free before use error.
`microbit_display_exception` frees the passed string immediately after `mp_hal_display_string` returns. `mp_hal_display_string` uses `microbit_display_scroll`, but does so without waiting.
Thus the passed string is freed while `microbit_display_scroll` is still using.

closed 2016-02-07 by dpgeorge