bbcmicrobit/micropython #199: Create a 1.0 release from the current code.

ntoll, 2016-02-08:
Github has a notion of a "release". We should do that for 1.0 and then sleep for 1000 years.

JoeGlancy, 2016-02-08:
Closer to now or the micro:bit's actual release?

sleep(((1000*365.24)+2.5)*24*60*60)

ntoll, 2016-02-08:
It's coming... it's coming... but these things take time and lots of people are putting in a huge amount of effort to make this happen. I'm hopeful we'll get good news imminently. :-)

JoeGlancy, 2016-02-08:
Hopefully :) I guess now is the time to wrap any other issues and get documentation perfected then?

dpgeorge, 2016-04-09:
We should tage "v1.0" ASAP.  What are the blockers?

markshannon, 2016-04-14:
No one has raised any objections, so go for it.

dpgeorge, 2016-04-17:
@ntoll what do you think about v1.0?  I think we should first address the following: #244, #250, #255, #257.  It would also be nice to have FS support #251 but that requires a bit more than just merging that PR (eg adding hexmain.py, lots of testing).

ntoll, 2016-04-17:
+1 - AFAICT we just need #250 solved by pulling from upstream. Once this is completed I'll update the web editor (pending BBC testing) and all the other tools.

dpgeorge, 2016-04-17:
@ntoll pulling from upstream is not straightforward: microbit version lives on a special branch upstream that needs rebasing first.... I might be able to do this by end-of-day.  If not I can just make a PR for the single change to fix the map resize.

ntoll, 2016-04-17:
FWIW my preference is that we take out time and do it properly. There's no deadline per-se so don't knock yourself out for end-of-today (go enjoy the sunshine!). :-)

ntoll, 2016-04-19:
I believe we now have everything to tag for 1.0. I'll defer to @dpgeorge for the final decision.

Once tagged I'll update all the tools and the online editor.

dpgeorge, 2016-04-19:
Ok then let us tag v1.0... but do we first want to add back the reset_presses() method (and reset_gestures()) for legacy scripts to work?

carlosperate, 2016-04-19:
If reverted back, how long would they be kept as "deprecated" functions? Normally breaking changes come with "major" versions, do we consider 1.0 to 1.1 to be a "major" update?

sparkslabs, 2016-04-19:
http://semver.org/ is quite a good basis for versioning. Whether you/anyone wants to use this or not, it's quite good for understanding **potential** expectations on version numbers. In practice I've yet to see many projects actually stick to that model "properly", but it's quite useful.

The short version of that is:

_Given a version number MAJOR.MINOR.PATCH, increment the:_
- _MAJOR version when you make incompatible API changes,_
- _MINOR version when you add functionality in a backwards-compatible manner, and_
- _PATCH version when you make backwards-compatible bug fixes._

As I say though, most projects that I've seen try to do this tend not to do this imperfectly.

JoeGlancy, 2016-04-19:
We should have codenames for each release (either major or minor, I'm not
sure).

ntoll, 2016-04-19:
If we have code names I suggest musical instruments as a naming convention, starting with "dijeridu" in honour of @dpgeorge. I also like "nose flute", "moog" and "dumdum" as future names. ;-)

JoeGlancy, 2016-04-19:
Hell yeah, those sound good :D

ntoll, 2016-04-19:
More seriously, I'd vote we should move forward and *not* reintroduce reset_presses(). Pre-1.0 API changes like this are allowed / expected. 

However, we should have a deprecation work-flow that we can point people at for the future. It's important to get this done properly in such a way that the community don't get annoyed / frustrated.

JoeGlancy, 2016-04-19:
Have a markdown changelog inside the repo which documents everything? Also,
we need somewhere where people can download the latest (stable & maybe
latest commit, using an online build robot?) micropython builds.

dpgeorge, 2016-04-19:
> More seriously, I'd vote we should move forward and not reintroduce reset_presses(). Pre-1.0 API changes like this are allowed / expected.

I agree with this.

> Have a markdown changelog inside the repo which documents everything?

That's what `git log` is for :)

> Also, we need somewhere where people can download the latest (stable & maybe latest commit, using an online build robot?) micropython builds

Other ports live at https://micropython.org/download and I can easily add microbit to that list.  The latest version is built twice a day.

JoeGlancy, 2016-04-19:
I meant a changelog as in added features, removed features, higher-level
stuff for people who just want to see what happened in simple English
(without having to look through a load of commits :).

whaleygeek, 2016-04-19:
'important to get this done properly in such a way that the community don't get annoyed / frustrated'

@ntoll I'm already annoyed/frustrated with the change. Although don't let that influence you. If this is the *only* breaking change to the API that was current on 22nd March 2016 when the mass deployment started, I'm sure we can live with it -- as long as you add a note in the docs against the new API methods explaining what the old methods were, why they were taken out, and how to convert a previously working program (now broken) back into a working program again. It's one change, so not worth loosing sleep over.

However, +10^yotta for getting the deprecation workflow defined, that will be really important, and things like allowing Mu to cache the .hex file elsewhere other than the micro:bit will ease some of the pain.

ntoll, 2016-04-19:
@whaleygeek +100000000 - the upshot of this breaking change is we need to add a note to the docs.

carlosperate, 2016-04-19:
Aren't we forgetting about the online editor? If people load their old scripts saved on the website they will have a hard time figuring out why they aren't working any more.

ntoll, 2016-04-19:
@carlosperate the BBC & others have been adding and then removing features from the live site since the deployment started. I guess it's something that the BBC don't mind (so much) although it is definitely something that should be flagged to them (which I've already done - pending a response).

carlosperate, 2016-04-19:
I wasn't aware they were also removing features and/or making breaking changes, if that's case then it sounds like the best course of action is to leave that deprecated functions outs for 1.0.

Is the documentation going to be tagged and stored in readthedocs as well? is there a "pre-first-tag" feature for read the docs? If, for whatever reason, somebody was using the old version and wanted to check the old documentation, would that be available online?

dpgeorge, 2016-04-20:
How about this for a plan of action:

- retroactively tag 0.9 as the version on 22nd March (launch date)
- add a dummy button.reset_presses() function that does nothing, and document it as deprecated
- do the same for accelerometer.reset_gestures()
- tag that as 1.0
- move forward!

carlosperate, 2016-04-20:
Sounds good, but could there be a non-intrusive way of alerting the users that the functions have been deprecated? (apart from the docs). In JavaScript I've seen libraries print on the console.out, which is quite useful when something unexpected gets the boot.

markshannon, 2016-04-20:
How about, as above but skipping steps 1-3 :wink: 
Seriously, let's just tag 1.0 and move on. We have plenty of new features to add.

Please don't add dummy functions that do nothing. How many 11 years know what "deprecated" means?

ntoll, 2016-04-21:
So, into this mix comes the BBC who told me yesterday that they won't be able to start the "test" process they require for the code editors to be updated until end of May/June. It means that while we may release new versions (and I hope we do), the BBC's test/release cycle is glacial, bureaucratic and manually driven so the earliest kids could get these changes is June.

**THIS IS UNACCEPTABLE** and I'm starting to push appropriate BBC buttons to undo this sorry state of affairs.

In terms of what we should do WRT 1.0. I agree with @dpgeorge that we should retroactively tag 0.9 but also agree with @markshannon that we should ignore 2-3 and move forward.

Does this all make sense? ;-)

markshannon, 2016-04-21:
> Does this all make sense?

The behaviour of BBC makes no sense whatsoever :confused: 
As to tagging versions, yes that makes perfect sense.

dpgeorge, 2016-04-21:
We (at least the 3 of us) are in agreement, great!

We should provide v0.9 binaries, somehow (within Mu?).  Also need a way for the user to inspect what version they have.  On other uPy boards you use `os.uname()`, for example:
```
>>> os.uname()
(sysname='pyboard', nodename='pyboard', release='1.7.0', version='v1.7-8-g1c0343f-dirty on 2016-04-18', machine='PYBv1.0 with STM32F405RG')
```

For microbit it would be something like:
```
>>> os.uname()
(sysname='microbit', nodename='microbit', release='0.9', version='v0.9-g1c0343f on 2016-04-18', machine='microbit with nRF51822')
```
That way the user can inspect the version themselves (or programatically).

Note that there's a difference between the uPy version (eg v1.7) and the microbit code itself (eg 0.9).

markshannon, 2016-05-03:
@dpgeorge Are you sure you're not imagining `os.uname`? :wink:
https://docs.micropython.org/en/latest/pyboard/library/os.html

I'm not sure where we should put the version number.
None `sys.implementation`, `sys.platform` or `sys.version_info` fit the bill.

`microbit.VERSION` or `microbit.version` perhaps?

dpgeorge, 2016-05-03:
os.uname() is implemented, even if it's not documented :)  It is the right place to put this information.

markshannon, 2016-05-03:
OK, `os.uname()` it is. It will conflict with the `os` module in https://github.com/bbcmicrobit/micropython/pull/251, but I've survived worse.

dpgeorge, 2016-05-03:
Since tagging is not a priority anymore we can merge #251 first, then do os.uname() later, then tag 1.0.

ntoll, 2016-05-04:
Can we hold on #251 for v1.0 please..? This will introduce an inordinate amount of BBC related bureaucracy that will hold things up in terms of being able to release the updated version of the editor. How about we release 1.0 pre-filesystem and then I can get the ball rolling for a 1.1 version containing the outstanding filesystem work. This could be quickly followed by a 1.2 containing the radio module and then perhaps 1.3 with speech/audio work.

For new features the BBC insist on testing themselves and this is based upon documentation we provide for them both in terms of the API and tutorials I've been writing. I've held off doing this for the filesystem work until we had a clear way forward.

markshannon, 2016-05-04:
"An inordinate amount of BBC related bureaucracy" sounds like a bad thing.
I don't much care whether the file system stuff is in 1.0 or 1.1.

I hope that adding a version number via `os.uname()` doesn't count as a "new feature".

ntoll, 2016-05-04:
Also, if it's undocumented, how will they know how to test it? ;-)

markshannon, 2016-05-13:
I'd like to make a 0.9 and 1.0 branch and then submit a minimal PR containing an `os` module with `uname` to both.
I would suggest branching 0.9 from c8016047521a1c5d81443fbc59f72600a20c0e67
and 1.0 from cb612964a158414e5110c4e18c0a9e53f6b95d03
Does that sound OK to everyone?

dpgeorge, 2016-05-13:
> I would suggest branching 0.9 from c801604

Yes, that's good, that's just before the official launch at LSE.

> and 1.0 from cb61296

How about merge #265 (soft reboot stuff), then add os module with uname(), then tag 1.0?

Moving from there, add the filesystem and tag 1.1?

Then radio support at 1.2?

JoeGlancy, 2016-05-13:
Does that mean we have a working MicroPython + Bluetooth stack then?

dpgeorge, 2016-05-13:
> Does that mean we have a working MicroPython + Bluetooth stack then?

No, unfortunately.  "Radio" support is simple peer-to-peer radio packets.  Much easier to use than BLE, but can only communicate between microbit.s

carlosperate, 2016-10-28:
So what the current plan with the versioning? we've been using 1.7 since April and there has been quite a few changes which are completely untracked (filesystem, radio, speech, pin refactoring, etc).

Additionally it would be quite convenient to have a way to identify the version directly from the hex file, some type of header or known location with or without markers? This way when people load old hex files with their scripts into editors we could use this information if required.

dpgeorge, 2016-11-03:
> we've been using 1.7 since April

That's the version of upstream MicroPython, which is different to the microbit firmware as a whole.

We did do a 0.9 but that's actually a branch not a tag.  And `os.uname()` does report the release as 1.0, but there's no branch or tag called "1.0".  We should really fix this.

nealmcb, 2019-10-10:
Can't this be closed (with congratulations!) based on the Sep 20, 2018 realease of 1.0.0 at https://github.com/bbcmicrobit/micropython/releases/tag/v1.0.0 ?

dpgeorge, 2021-09-06:
Yes!  It can be closed.

closed 2021-09-06 by dpgeorge