bbcmicrobit/micropython #204: Contributed drivers

JoeGlancy, 2016-02-11:
Many people will (probably) want to interface a variety of external chips and ICs to the micro:bit, and have some way to access those through Python. However, some (if not most) of those chips will not be "mainstream" enough to be always built into micropython, taking into account the restricted amount of flash we have. So, what I'm proposing is a `contrib` folder inside `lib`, for the sole purpose of storing drivers for whatever people have made them for (e.g: a multichannel LED driver IC). Each of these will be disabled by default and not compiled, but they will all have the ability to be enabled through something like a header file with definitions stating what is to be built and what's not. Also, I propose a new module for micropython, called `contrib`, which will contain an interface to the Python API for each of these IC drivers, also enabled at build-time. There can also be a contrib folder inside source/microbit which is where the actual Python APIs for each IC live, whereas source/microbit/modcontrib.cpp (mentioned above) will bring those all into the module itself. For example, an MY9221 LED driver might have the following layout:
 - `source/lib/contrib/my9221.c` - The C API which uses the microbit's pin I/O to interface with the chip
 - `source/lib/contrib/my9221.h` - The header file for the C API
 - `source/microbit/contrib/microbitmy9221.cpp`- A micropython API interfacing with the C API

How does it sound? It would be ideal to have an interface similar to this as it might make it easier to interface your own drivers with micropython, perhaps using templates for things like the Python part.

dpgeorge, 2016-02-11:
If possible, drivers should be written in Python.  Then there's no compiling, it just runs!

On the microbit there's the problem that we don't (yet) have a filesystem so you can't import 3rd party Python scripts.  Also, since there's not much RAM you likely won't be able to import more than one, unless the drivers are very small.

An alternative is to use frozen (precompiled) bytecode.  Then you can still write the driver in Python (and have it be compatible with other MicroPython boards) but it doesn't take up any RAM, and doesn't need a filesystem.  It does though require you to recompile the firmware.

JoeGlancy, 2016-02-11:
With the contributed drivers, you could simply rebuild it and compile it in
when you want, plus I guess it could be smaller than the bytecode? A
microsd card could be connected to the uBit, but that would probably
require external hardware and would be fiddly. Plus, as you say, imported
libraries would/could take up a hefty amount of RAM, so bytecode would be
more ideal if it's compiled in.

gbaman, 2016-02-11:
Are we sure Python could actually keep up with an I2C device for example? Is it fast enough? Assumed would need to be written down at C++ level to be able to work fast enough?
It is an interesting one, because at the same time it is pushing it to ask the kids to build their own firmwares, or even the teachers. Yet, we don't have enough space on the device to include the libraries...
Is there any way for example Mu could add extra libraries (written in C++ and compiled) onto the end of the hex blob? Mu could have those ready to put on the end if it detects it being imported?
Am I right in saying anyway though that the modules wouldn't be imported into RAM unless specifically imported in Python?

carlosperate, 2016-02-11:
> Are we sure Python could actually keep up with an I2C device for example? Is it fast enough? Assumed would need to be written down at C++ level to be able to work fast enough?

The peripheral drivers would still be in C++, but the "IC driver logic" could be written in Python.

>An alternative is to use frozen (precompiled) bytecode. Then you can still write the driver in Python (and have it be compatible with other MicroPython boards) but it doesn't take up any RAM, and doesn't need a filesystem. It does though require you to recompile the firmware.

@dpgeorge Similar to the question by @gbaman, if the drivers python code is "hexified", couldn't we just have a module that, per import, attaches the hex blob to the end of the python script invisibly to the user?

JoeGlancy, 2016-02-11:
@gbaman remember, not *everyone* uses Mu :). @carlosperate another thing to consider is the fact that a lot of existing driver code is already in C/C++, which would make porting over drivers easier (however I agree that the logic could be implemented in Python, but the uBit has 16KB of RAM (I believe), which with a 4kb driver (which could happen, hexified or not) could be exhausted quickly.

whaleygeek, 2016-02-11:
You're all missing the point here with respect to speed.

The I2C clock is driven from the master end (the micro:bit), you can run it
down to DC (and I have done on many platforms). You can run the clock as
slow as you need to, as the slave will use the SCL clock timing as
delivered by the master end.

For example I have a bit-bashed I2C interface that I use on some platforms,
that is written directly in python. It works fine, all the time, even when
I stop at breakpoints in the code with no hardware assisted peripheral
support. The same is true of SPI.

The only time it becomes an issue is if you want the micro:bit to be the
slave end, in which case it has to keep up with the master SCL clock -
although a full slave implementation can use I2C clock stretching to slow
the bus down to the rate that it can cope with.

There are peripherals on the ARM chip that do all the timing for you with
hardware assist. So you could write it in C/C++, or you could write it in
Python. You choose!

D

deshipu, 2016-02-11:
An easy way to add more modules would certainly be great, but of course we are limited here. I think that even doing something similar that the first ESP8266 port had -- putting your python code in the ``scripts`` directory and recompiling the firmware -- would be helpful. I suppose with some work it could also be hexlified and appended to the firmware together with the program code, but I assume this is more work, and it's nothing I can do.

Writing most of those drivers in Python should be no problem at all. All the I²C, SPI, UART, etc. communication is handled by hardware anyways, as well as timers and PWM. You would probably need C++ for things like 1Wire, WS2812 or HT11/22 code, but that should be relatively small. All the high-level libraries for LCD screens, 7-segment displays, thermal printers, distance sensors, TV remotes, radio and WiFi modules, etc. should probably be written in Python.

JoeGlancy, 2016-02-12:
Limited in what sense, flash memory space?

deshipu, 2016-02-12:
First, Microbit is limited in how you can communicate with it. Because the communication is handled by a separate chip, it's impossible, without reprogramming that chip, to do anything else than sending a .hex file for flashing. In particular, it's not possible to have the Microbit visible as an USB disk to which you can save files, like on other Micropython-based boards.

Second, the microbit has limits both in the size of the memory for the program, and the memory for data. Features such as a filesystem come at the cost of the programs you could run on it. I don't know the exact numbers here, but it's not made of rubber.

And most importantly, it all takes work. Some of this is easier, some harder. It's the kind of work that I can't do yet. Not sure how many volunteers are there willing to work on it. Perhaps some of the work on the ESP8266 port will make this easier. Even then, I suppose that the person who actually implements it gets to decide how it's done. It could be you! I will certainly not go around telling people what they should work on (unless they ask).

JoeGlancy, 2016-02-13:
Space and device-to-other-device communications aren't too important. The idea is to have *only* the drivers you want compiled directly in at build-time, and have them available through a specific Python module. The only downside (if you can call it that) is that the other drivers aren't available, only the ones you want are, however if you did need them you could simply recompile them back in. The ones you don't need will have no trace in the resulting binary blob, and so will not have any effect on the size.
Work wise, it's just designed to provide an interface for whoever to add their own drivers for others to use. The base of it is just a module, which will have a simple template for people to interface the low-level driver with the high(er)-level Python module. If people are going to want to expand their micro:bit's functionality with, for example, a motor driver for a robot or a LED driver IC with its own strange serial interface (as some do), sure they can write their own Python driver, but some might want to just have it there for instant use.
One rather large issue is that they will need the micro:bit micropython development/build environment, but once the DAL is open sourced (hopefully soon) this shouldn't be the hardest thing to achieve. Maybe someone might host an online micropython build service? Who knows :)

markshannon, 2016-10-11:
@c0d3st0rm do you thinks this needs further action? I'd like to close it otherwise.

markshannon, 2016-11-07:
No further action requested, so closing.

closed 2016-11-07 by markshannon