bbcmicrobit/micropython #208: Image instantiation BROKEN :-(

ntoll, 2016-02-14:
To recreate, with the latest version of MicroPython from master:

```
from microbit import *

i = Image("01010:"
          "01010:"
          "01010:"
          "11111:"
          "01110:")
display.show(i)
```

Should display a very dim sailing ship. It currently displays random pixels at full brightness. If I change the brightness to `9`, it works perfectly. It appears, any value not `0` or `9` causes the problematic output.

Has the recent update to the DAL caused this problem..? (E.g. pin4 and pin5 swapping capabilities.)

Would it be possible to fix this as a matter of some urgency. The BBC are currently testing MicroPython and if they discover this is broken it'll create a bunch of bureaucratic delays I'd rather we not face.

@dpgeorge @markshannon any ideas..?

markshannon, 2016-02-14:
Works perfectly on my build (without the recent update), so it probably is a result of the latest DAL update.
Possibly a timer issue. I'll look into this evening.

ntoll, 2016-02-14:
Cheers @markshannon! I'm writing up a bunch of lessons to be delivered to the BBC tomorrow for editing / testing by the BBC. Lesson 3 is all about images. ;-)

(They'll also be part of the docs for MicroPython.)

markshannon, 2016-02-14:
BTW, which version of the hardware?

ntoll, 2016-02-14:
I've tested it on both old and new versions of the hardware. It behaves (errors) the same no matter the version of the hardware you're using.

ntoll, 2016-02-14:
@markshannon both of the boards you have should display the same symptoms.

ntoll, 2016-02-14:
I've just checked out and built from 240c6173c036d9a5683e9a25b8c8adc6ad291313 (the revision before the most recent DAL update). It works perfectly. Since there are only two subsequent commits I hope it'll be relatively easy to spot what has changed.

BTW, @markshannon the board I gave you when last we met is the latest revision of the hardware.

ntoll, 2016-02-14:
Unsurprisingly, it's broken ``love`` ;-) Such a sad thing to discover on Valentine's day.

closed 2016-02-14 by dpgeorge