bbcmicrobit/micropython #210: Simple radio module?

ntoll, 2016-02-14:
See ble-services/MicroBitRadio.cpp in the DAL.

JoeGlancy, 2016-02-14:
`source/ble-services/MicroBitRadioDatagram.cpp` might be more suited to this, would it not?

finneyj, 2016-02-18:
Folks - please be aware that the BBC child protection team are currently reviewing this low level radio API to ensure were not putting kids at any risk here (I know, but these guys want to ensure there are no privacy concerns etc), and have requested that all languages refrain from making any public APIs visible until the review is complete...

zeth, 2016-03-10:
There is a nice API now visible here: 
https://github.com/Microsoft/yelm/blob/master/libs/microbit-radio/radio.ts

So can we assume from that the review finished?

zeth, 2016-03-10:
Ignore last, it is answered here:
https://github.com/bbcmicrobit/micropython/issues/71

markshannon, 2016-11-07:
We have a radio module now.

closed 2016-11-07 by markshannon