bbcmicrobit/micropython #214: Please clarify get_gestures behaviour

ntoll, 2016-02-16:
It appears from practical use (by the BBC) that only detected gestures (triggered by, say `current_gesture()`) find their way into the tuple returned by `get_gestures`. Put simply, you need to explicitly detect a gesture for it to be added to the tuple.

My understanding was there is some sort of ongoing background listener (i.e. `accelerometer_listener` in the C code) and explicit detection wouldn't be needed.

Have I described things correctly? If not, is this a bug and can we fix it? I need to clarify the documentation and advise the BBC. 

@dpgeorge..?

dpgeorge, 2016-02-16:
Try the following:
```python
while 1:
    print(accelerometer.get_gestures())
    sleep(100)
```
Then wave the microbit around, put it at different orientations, etc, and you'll see the list getting populated.

We discussed this (but I guess it wasn't clear): you need to call some accelerometer function (eg get_x, get_values, etc) for the listener to update.  There is no true background process doing the gesture detection (that would require a lot of power, and I2C contention issues, as you'd need to constantly use the I2C bus to query the accelerometer, say every 50ms).

I'm not sure how it works with TD and the other languages...

ntoll, 2016-02-16:
@dpgeorge thank you! :-)

closed 2016-02-16 by ntoll

markshannon, 2016-02-16:
Also, storing gestures takes memory. The underlying data structure is very efficient, but creation of the result tuple might fail with several hundred gestures.